sharpie7/circuitjs1 #439: 555 square wave example

albe43, 2020-04-26:
555 square wave example.  Why does the example show 240 Hz?  Should it not be 2.4 Hz?
Thanks

pfalstad, 2020-04-26:
Why should it be 2.4 Hz?

albe43, 2020-05-03:
Several other 555 calculators show 2.4 Hz

pfalstad, 2020-05-03:
Which ones?

albe43, 2020-05-04:
Several other 555 calculators show 2.4Hz For 10K , 1meg , 300nf

albe43, 2020-05-04:
They,re all probably wrong and you,re right

closed 2020-05-04 by albe43

pfalstad, 2020-05-04:
You mean like this one?

http://www.ohmslawcalculator.com/555-astable-calculator

That's not the same circuit as the one on my page.

If you use this circuit, the calculators will work: http://tinyurl.com/ydfvtrqj  I don't use that circuit because it doesn't have a 50% duty cycle.

albe43, 2020-05-05:
Maybe I’m not reading the values right.  This is what it look like to me: 10k, 1 meg, & 300nanoF.  I plugged these values into the first calculator that you suggested and I got nearly 2.4 hz and nearly 50% duty cycle.

I am new to electronics so, if you please, tell me how your circuit is different than the other sites.

Thanks,

Bill

 

http://www.ohmslawcalculator.com/555-astable-calculator

pfalstad, 2020-05-06:
So the square wave example on my site is this one: https://www.falstad.com/circuit/e-555square.html

It's just a different circuit than the one on the calculator page.  On the calculator page, the output isn't connected to anything.  On my site, the output is connected back to the trigger input through a 10k resistor.  And there are various other differences.

If you want to use the 555 calculator, then use this circuit instead:

http://tinyurl.com/ydfvtrqj

This one matches the circuit on the calculator page.

albe43, 2020-05-07:
I now understand the definition of a “square wave.”  Any wave that is not, strictly, 50% duty cycle is not a “square wave.”  It must be strictly symmetrical to be a “square wave.”  Everything else is a wave with sharp leading and sharp ending, not “square” .  You specified a “square wave generator.”  As a newcomer, I was not aware of this designation.  I didn’t pick this up from your example text.  Now knowing this, I think that I can go out and win a few beers.

 

Thanks again,

Bill

 

http://tinyurl.com/ydfvtrqj