sharpie7/circuitjs1 #440: Compile problem

popkingcorn, 2020-04-29:
Hello,
when i try to compile a project i get an error:
[ERROR] Hint: Check that your module inherits 'com.google.gwt.core.Core' either directly or indirectly (most often by inheriting module 'com.google.gwt.user.User')

Changing GWT SDK version to 2.8.1 did't help.
It's very important for me to fix this, hope you can help...

JFalstad, 2020-04-29:
What JRE are you using?  I am using jdk1.8.0_131.

It looks like you need GWT SDK 2.8.2 to use Java 9 or later: http://www.gwtproject.org/release-notes.html#Release_Notes_2_8_2

popkingcorn, 2020-04-29:
I had JDK 13 and Java 8 Update 241 installed on my pc.
In project explorer in Eclipse JRE System Library set as J2SE-1.5

I have installed jdk1.8.0_131 and it's fixed my issue!
I'm very appreciate your help. Thanks a lot.

closed 2020-04-29 by popkingcorn

chaojian-zhang, 2020-12-21:
I am having the same issue with Java version "1.8.0_261".

I solved this problem by following advice on this discussion (https://stackoverflow.com/questions/11691575/how-do-upgrade-to-gwt-2-5-in-eclipse) and replaced GWT to version 2.9.0.

Here is my project setup:

[image]