sharpie7/circuitjs1 #441: Incorrect behavior of LEDs

KonstantinDi, 2020-04-29:
Hi, it is not possible to set the characteristics of a normal white led type 5370, 0.5 watt power, voltage drop of 3.2 volts, maximum current of 150mA. I tried to change the parameters in the custom selection section, but it turns out strange. I put the current source at 150mA, and watch the voltage drop, picked up the parameters so that it became 3.2 V, then change the current source to 50mA. In reality, the voltage drop should be 2.9 V, but in the program it is for some reason 1.7 V. As I did not select the parameters in the custom section, I could not get an analog of a white led. Although 2 years ago, when I used this program, everything was close to reality.

pfalstad, 2020-04-29:
Here you go

http://tinyurl.com/ybvywm8n

I found this datasheet showing a voltage drop of 2.2 V at 10mA so I fit the model to that.

https://datasheet.octopart.com/5370T5LC-CML-Innovative-Technologies-datasheet-7824027.pdf

The datasheet doesn't have an IV curve or anything like that so there isn't a lot of information I can use to make it more accurate.

KonstantinDi, 2020-05-02:
Hi, thanks, apparently this is wrong datasheet. I found one that describes the correct behavior of 5730 LEDs. Take a look, please. Page 6, there is a curve of IV. I hope it will help. I tried to simulate this behavior of led, but i failed. If you try to simulate the correct behavior of the LED, it will be very useful. Thanks for the answers!
[https://image.micros.com.pl/_dane_techniczne_auto/olb.5730.4000k.pdf](url)

Leds what i mean looks like that [https://micros.com.pl/en/catalog/082061075/smd-leds-power-05w-5730](url), it`s just an example, may be will help too.

pfalstad, 2020-05-02:
How about this one: http://tinyurl.com/ycunot9y

One of these days I will make a worksheet to make it easy to generate these models.

KonstantinDi, 2020-05-03:
Wow, that’s what I need, thank you very much! But how did you get the right result? Worksheet, you mean a special method, or just a table with parameters. Sorry, but English is not my native language, and I could not understand what you meant by the word worksheet.

pfalstad, 2020-05-03:
I meant something like this: http://www.falstad.com/circuit/diodecalc.html

KonstantinDi, 2020-05-12:
That`s realy great! Thanks! This tool http://www.falstad.com/circuit/diodecalc.html is available, only from here (your comment), or it`s possiable by refer on your site http://www.falstad.com?

closed 2020-05-16 by sharpie7