sharpie7/circuitjs1 #442: Error Creating Subcircuit

AndrewIsOffline, 2020-05-03:
I'm getting an uncaught exception error when I try to draw a subcircuit. The subcircuit creation works just fine, but doesn't draw anything.
Attached are the error log and the circuit.
[www.falstad.com-1588549067553.log](https://github.com/sharpie7/circuitjs1/files/4572254/www.falstad.com-1588549067553.log)
[circuit-20200503-1942.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4572256/circuit-20200503-1942.circuitjs.txt)

pfalstad, 2020-05-04:
The simulator doesn't seem to like the fact that I1 goes straight through to O11 without being connected to any components.  If you remove these terminals, or put two inverters in between, it works.  I'll make a note to fix this or at least give a reasonable error message.

sharpie7, 2020-08-02:
Fixed

closed 2020-08-02 by sharpie7