sharpie7/circuitjs1 #444: Feature request: activated (blown) fuse simulation

jamison-io, 2020-05-17:
Is it possible to simulate failure modes such as a fuse blowing from too much current? If not, I would like to request that as a feature. I found one thread from 2016 that mentions it, but I do not believe anything came of it.

pfalstad, 2020-05-17:
Yes, there's a fuse under "Passive Components"

jamison-io, 2020-05-17:
Yes, I saw that, thank you. When I increased the amperage through it to a very high level, it didn’t blow.  Are you saying that the simulation will show the fuse activated and the circuit will be interrupted at that point?

pfalstad, 2020-05-17:
Yes.  You may have to adjust the fuse parameters to match what fuse you are trying to model.  How many amps are we talking about?

jamison-io, 2020-05-17:
I see my mistake now - I was thinking an amperage source represented an amperage draw. I swapped it out for a lamp and blew the fuse!

closed 2020-05-17 by jamison-io

mattehalen, 2022-02-05:
How would I modify the parameters to replicate a 16A fuse or a 20A or a 63A? If I use Ohms low and calculate how big the resistance should be if I have 230V and 16A I get that the resistance should be 14,375ohm but If I but that in the fuse parameter it "blows" the fuse at around 4A. What do I do wrong?

pfalstad, 2022-02-05:
The parameter you want to change is I2t.  The simplest way to get that is to find the datasheet for the fuse you want to use, or google for a similar fuse.

I found this table: https://www.jameco.com/Jameco/Products/ProdDS/1711648.pdf

For a 15A/250V fuse they have resistance = 0.0049, i2t = 245.
for 20A, they have 0.0033, 575.