sharpie7/circuitjs1 #445: Quality of Life: When leaving the page, ask user to confirm if they want to save circuit.

tejashah88, 2020-05-22:
I've been working with this circuit simulator and I love it so far! The only issue is that if I accidentally reload the page or close it, all my progress is lost and I have to end up making everything from scratch again. I'd like it if there was an alert dialog that asks me if I want to save my circuit to a text file or cancel reloading/leaving the webpage.

pfalstad, 2020-05-22:
You can do File->Recover Auto Save.

tejashah88, 2020-05-23:
@pfalstad Thank you for the quick reply! That definitely helps. I still think that popup would be helpful, since most people may not know about the "Recover Auto Save" (like me) and become frustrated about building more complex circuits.

closed 2020-12-05 by sharpie7