sharpie7/circuitjs1 #446: Protections sous circuitjs1

Fvi589, 2020-05-25:
Bonjour,
Je suis professeur de Génie Électrique dans un lycée français, et je trouve votre logiciel très intéressant pour montrer certaines problématiques à mes étudiants en BTS Electrotechnique. 
J'aime le côté OPENSOURCE et le fait que cela soit très convivial d'utilisation.
J'ai trouvé le fusible dans les éléments de protection d'un circuit. J'ai regardé le code JAVA afin de comprendre le fonctionnement, basé sur le i2t, mais en passant en simulation, j'ai des doutes car pour un i2t de 298 A2s, ca déclenche instantanément, mais à 300 A2s, le fusible ne fond jamais.
Ma question est de savoir, si vous pensez à l'avenir intégrer les disjoncteurs dans la librairie ?
Je suis conscient que c'est de l'OpenSource, mais est ce que cela est envisageable ?
En tout cas, félicitations !!!
En vous remerciant par avance,
Cordialement,
Franck VIARD
-----------------------------------------------------------------------------------------------------
traduction Google :
Hello,
I am a professor of Electrical Engineering in a French high school, and I find your software very interesting to show certain problems to my students in BTS Electrotechnique. I like the OPENSOURCE side.
I found the fuse in the protection elements of a circuit. I looked at the JAVA code in order to understand the operation, based on the i2t, but passing in simulation, I have doubts because for an i2t of 298 A2s, it triggers instantly, but at 300 A2s, the fuse blows never.
My question is, if you are thinking of integrating circuit breakers in the library in the future?
I am aware that it is OpenSource, but is it possible?
In any case, congratulations !!!
Thanking you in advance,
Best regard,
Franck VIARD

pfalstad, 2020-05-25:
The fuse is working for me, at 298 or 300.  Could you send a sample circuit showing the problem?  You can use File->Export As Link.

By the way, do your students have any problem with using the simulator in English?

Le fusible fonctionne pour moi, à 298 ou 300. Pourriez-vous envoyer un exemple de circuit montrant le problème? Vous pouvez utiliser File->Export As Link.

Soit dit en passant, vos élèves ont-ils des problèmes avec l'utilisation du simulateur en anglais?

pfalstad, 2020-05-25:
We could certainly add a circuit breaker at some point.  It wouldn't be hard.

Nous pourrions certainement ajouter un disjoncteur à un moment donné. Ce ne serait pas difficile.

Fvi589, 2020-05-26:
Bonjour M FALSTAD,

Je vous remercie de votre réponse. J’ai refait le montage et je trouve la même chose. 
J’ai mis la base de temps à 1 sec.
Pour ma part, j’ai compris la fusion du fusible, tel que :
I2t = 300 A2s, donc si j’ai 10A qui traverse le fusible, i2 = 100A2, il doit tenir 3sec.
Le montage :
http://lushprojects.com/circuitjs/circuitjs.html?cct=$+1+1+10.20027730826997+50+5+43%0Av+240+304+240+144+0+0+40+10+0+0+0.5%0Ar+368+208+368+304+0+1%0A404+368+144+368+208+0+0.000001+298+0+true%0A370+240+144+368+144+1+0%0Aw+240+304+368+304+0%0Ag+240+304+240+336+0%0A38+2+0+100+400+I2t%0A

Pour I2t = 298, déclenchement instantané
Pour I2t = 300, pas de déclenchement 

Je pense utiliser de plus en plus LushProject, avec mes étudiants. Ils l’apprécient également ; voir avec des couleurs, ce qui se passe dans les circuits (diodes, etc …) est une très bonne idée, en autre. 
Comme vous dites, la simulation ne remplace pas la réalité, mais cela leur permet de mieux appréhender les phénomènes physiques. Ils sont en BTS, donc en France, cela fait BAC+2, et cela fait partie de leurs compétences, de comprendre l’anglais, donc pas de problèmes.
En BTS Electrotechnique, on fait surtout des montages de puissance, mais la librairie Logic est très appréciable également.
Je pourrais contribuer à vos développements, de par mon parcours professionnel, j’ai déjà fait du développement informatique, et je pourrais sans doute traduire en français, les termes anglais.
 J’ai regardé avec d’autres langues, si j’ai bien compris, il suffit de traduire les termes anglais ?
Locale_xx.txt donc locale_fr.txt pour le français.
Je pourrais essayer, voire avec le prof d’anglais du Lycée des BTS.
Les composants qui pourraient être pertinent de rajouter ou que je n’ai pas trouvés (je m’excuse, je mets tout ce qui me passe par la tête 😊) :
-	Disjoncteur (calibre, réglage thermique, magnétique, pouvoir de coupure)
-	Différentiel (DDR)
-	Thyristor
-	Contacteur (même s’il y a déjà « relais »)
-	Pince ampèremétrique
-	Impédancemètre
-	Mesure du déphasage
-	Etc … 😊
En vous félicitant à nouveau du travail accompli,
Je reste à votre disposition,
Cordialement,
Franck VIARD (franck.viard@ac-dijon.fr)
PS : je vous laisse mon adresse mail, cela serait sans doute plus simple pour échanger 
----------------------------------------------------------------------------------------------
Hi M FALSTAD,

Thank you for your reply. I did the editing again and I find the same thing.
I set the time base to 1 sec.
For my part, I understood the melting of the fuse, such as:
I2t = 300 A2s, so if I have 10A crossing the fuse, i2 = 100A2, it must hold 3sec.
Mounting :
http://lushprojects.com/circuitjs/circuitjs.html?cct=$+1+1+10.20027730826997+50+5+43%0Av+240+304+240+144+0+0+40+10+0+ 0 + 0.5% 0Ar + 368 + 208 + 368 + 304 + 0 + 1% 0A404 + 368 + 144 + 368 + 208 + 0 + 0.000001 + 298 + 0 + true% 0A370 + 240 + 144 + 368 + 144 + 1 + 0% 0Aw + 240 + 304 + 368 + 304 + 0% 0Ag + 240 + 304 + 240 + 336 + 0% 0A38 + 2 + 0 + 100 + 400 + I2t% 0A

For I2t = 298, instantaneous tripping
For I2t = 300, no tripping

I plan to use LushProject more and more, with my students. They also like it; to see with colors, what happens in the circuits (diodes, etc…) is a very good idea, in other.
As you say, simulation does not replace reality, but it allows them to better understand physical phenomena. They are in BTS, so in France, that's BAC + 2, and it's part of their skills, to understand English, so no problems.
In BTS Electrotechnique, we mainly do power mounts, but the Logic library is also very appreciable.
I could contribute to your developments, from my professional background, I have already done IT development, and I could probably translate into English, the English terms.
 I looked with other languages, if I understood correctly, is it enough to translate the English terms?
Locale_xx.txt therefore locale_fr.txt for French.
I could try, even with the English teacher of the Lycée des BTS.
The components which could be relevant to add or which I did not find (I apologize, I put everything that goes through my head 😊):
- Circuit breaker (rating, thermal, magnetic adjustment, breaking capacity)
- Differential (DDR)
- Thyristor
- Contactor (even if there is already "relay")
- Current clamp
- Impedance meter
- Phase shift measurement
- etc … 😊
Once again congratulating you on the work accomplished,
I remain at your disposal,
Best regards,
Franck VIARD (franck.viard@ac-dijon.fr)
PS: I leave you my email address, it would probably be easier to exchange

Fvi589, 2020-05-26:
Je commence à traduire en français le contenu de locale_fr.txt

ENJOY !!! 😊

pfalstad, 2020-05-26:
OK, I see.  A timestep of 1 second is really huge.  I would pick something closer to the default value of 5 usec.

When implementing the fuse, I didn't really have a model for heat dissipation, so I assumed that a fuse can dissipate all its heat in 3 seconds.  So if it would take 3 or more seconds for a fuse to blow, it won't blow at all.  That seemed realistic to me, because fuses are supposed to blow fast.

With an I2t of 298, the fuse doesn't actually blow instantaneously.  It takes 7 seconds.  It just seems instantaneous because of the really long timestep.

A French translation would be great!  Yes, you just need to use one of the files like locale_de.txt and replace the German with French.

OK je vois. Un timestep de 1 seconde est vraiment énorme. Je choisirais quelque chose de plus proche de la valeur par défaut de 5 usec.

Lors de la mise en œuvre du fusible, je n'avais pas vraiment de modèle de dissipation thermique, j'ai donc supposé qu'un fusible pouvait dissiper toute sa chaleur en 3 secondes. Donc, si un fusible met 3 secondes ou plus à sauter, il ne sautera pas du tout. Cela m'a paru réaliste, car les fusibles sont censés sauter rapidement.

Avec un I2t de 298, le fusible ne saute pas instantanément. Cela prend 7 secondes. Cela semble juste instantané en raison du très long timestep.

Une traduction en français serait super! Oui, il vous suffit d'utiliser l'un des fichiers comme locale_de.txt et de remplacer l'allemand par le français.

pfalstad, 2020-05-26:
A circuit breaker can be implemented using a subcircuit like so:

http://tinyurl.com/ybhqelsb

example: http://tinyurl.com/y7gk4vrn

Fvi589, 2020-05-27:
[lushproject.pdf](https://github.com/sharpie7/circuitjs1/files/4689670/lushproject.pdf)

pfalstad, 2020-05-28:
Hmm, I'm not sure about the fuses.  This simulator is not intended to be like spice.  So we're not necessarily trying to match the behavior of specific components very precisely.  We generally like to implement idealized models.

Also these curves are not enough by themselves.  What if the current is 10 A for a short time, then goes back down to 1A, then up again?  It's not clear from the melting curves what to do with that.  We still need a model.  You mentioned RMS current.  What if it's not sinusoidal?  How do we compute the average?  Etc

This information could help me improve the model, though.  Clearly 3 seconds is too short.  Do you have a link?  These pictures are too small.  I'm sure I could make the fuse more realistic.

Hmm, je ne suis pas sûr des fusibles. Ce simulateur n'est pas destiné à être comme Spice. Nous n'essayons donc pas nécessairement de faire correspondre très précisément le comportement de composants spécifiques. Nous aimons généralement mettre en œuvre des modèles idéalisés.

De plus, ces courbes ne suffisent pas en elles-mêmes. Et si le courant est de 10 A pendant une courte période, puis redescend à 1A, puis de nouveau? Les courbes de fusion ne montrent pas clairement quoi faire avec cela. Nous avons encore besoin d'un modèle. Vous avez mentionné le courant RMS. Et si ce n'est pas sinusoïdal? Comment calcule-t-on la moyenne? Etc

Ces informations pourraient cependant m'aider à améliorer le modèle. Il est clair que 3 secondes est trop court. avez vous un lien? Ces images sont trop petites. Je suis sûr que je pourrais rendre le fusible plus réaliste.

Fvi589, 2020-05-28:
P24
[La_protection_des_circuits_et_des_person.pdf](https://github.com/sharpie7/circuitjs1/files/4694210/La_protection_des_circuits_et_des_person.pdf)

Fvi589, 2020-05-28:
Pour calculer la valeur efficace (RMS) ou la 
[image]
valeur moyenne d'un signal :

Fvi589, 2020-05-28:
Je suis d’accord avec vous, moi non plus je ne cherche pas à ce que les composants répondent exactement comme dans la réalité. 
C’est surtout l’ordre de grandeur des temps de déclenchement qui m’intéresse. Je peux essayer de voir à créer des allures génériques, que ce soit pour les fusibles, les disjoncteurs, les relais thermiques. 
Pour mon utilisation, je serai pleinement satisfait.
------------------------------------------
I agree with you, neither am I trying to make the components respond exactly as they do in reality.
It is above all the order of magnitude of the trigger times that interests me. I can try to see to create generic allures, whether for fuses, circuit breakers, thermal relays.
For my use, I will be fully satisfied.

Fvi589, 2020-05-28:
J'avance dans la traduction, mais c'est long !! 200/1000 :)

I'm advancing in the translation, but it's long !! :)

Fvi589, 2020-06-02:
Existe-t-il un document à suivre afin d’installer et de compiler facilement circuitjs en opensource svp ? 
Je voudrais commencer à développer des composants, et les mettre à disposition à la communauté. 
J’en suis à la moitié de la traduction en français de locale_fr.txt

Is there a document to follow in order to easily install and compile circuitjs in open source please?
I would like to start developing components, and make them available to the community.
I'm halfway through the French translation of locale_fr.txt

pfalstad, 2020-06-02:
Yes it's in the readme here: https://github.com/pfalstad/circuitjs1/blob/master/README.md

Fvi589, 2020-06-04:
J’aurais besoin d’aide !!! 😉
J’ai suivi les recommandations du fichier README.md mais je ne suis pas un PRO d’éclipse et des makefile en Java. Je connais plus les applis en C !!!
Je n’arrive pas à compiler l’appli, mais sans doute, je dois commettre des erreurs.
 
J’ai installé wampserver, car si j’ai bien compris, cela génère une appli web.
Est-ce que quelqu’un pourrait m’expliquer comment générer une appli (l’idéal en français, si possible), car je souhaiterai vraiment développer de nouveaux composants, mais il faut que je puisse tester pour voir ce que cela donne.
Merci d’avance !!!
I will need help !!! 😉
I followed the recommendations of the README.md file but I am not an eclipse and makefile PRO in Java. I no longer know the apps in C !!!
I can't compile the app, but probably I have to make mistakes.
 
I installed wampserver, because if I understood correctly, this generates a web app.
Could someone explain to me how to generate an app (ideal in French, if possible), since I would really like to develop new components, but I need to be able to test to see what it looks like.
Thank you in advance !!!
[lushproject.pdf](https://github.com/sharpie7/circuitjs1/files/4729264/lushproject.pdf)

pfalstad, 2020-06-04:
You don't need wampserver.  At least, I don't use it.

What is going on in "war"?  Click on it, expand it, see what file has an error, and what is the error.

Fvi589, 2020-06-05:
[image]
j'ai ce message d'erreur

Fvi589, 2020-06-05:
[image]

pfalstad, 2020-06-05:
Quel JRE avez-vous? J'ai jdk1.8.0_131.

Vous avez besoin de GWT SDK 2.8.2 si vous avez Java 9 ou supérieure: http://www.gwtproject.org/release-notes.html#Release_Notes_2_8_2

Fvi589, 2020-06-06:
J'abandonne, je laisse tomber. Dommage, car je voulais développer des composants. 
Je finis la traduction en français, cette semaine et je vous l'envoie.
------------------------------
[Uploading circuitjs1-master.zip…]()
[lushproject.pdf](https://github.com/sharpie7/circuitjs1/files/4739973/lushproject.pdf)

I give up ! Too bad, because I wanted to develop components.
I'm finishing the French translation this week and I'm sending it to you.

Fvi589, 2020-06-06:
[image]

pfalstad, 2020-06-06:
I don't blame you.  Eclipse can be a real hassle.

If you've installed jdk1.8.0_131, you may need to switch JDK's in eclipse by doing this: https://stackoverflow.com/questions/13635563/setting-jdk-in-eclipse

Fvi589, 2020-06-09:
[locale_fr.txt](https://github.com/sharpie7/circuitjs1/files/4753186/locale_fr.txt)
Je viens de finir la traduction. 
N'y a t il pas un autre moyen de compiler en java le build, l'application, sans passer par Eclipse. J'ai essayé avec Intellij, ca ne marche pas non plus, c'est gwt qui pose problème. Car je voudrais vraiment développer des composants.
---------------
I just finished the translation. 
Is there not another way to compile in java, the application, without Eclipse. I tried with Intellij, it does not work either, it is gwt which poses problem. Because I would really like to develop components.

pfalstad, 2020-06-09:
Merci beaucoup!

Regardez: http://www.falstad.com/circuit/circuitjs.html?lang=fr

Je ne sais pas un autre moyen de compiler..  désolé

pfalstad, 2020-06-09:
I noticed you used almost no accents..  Example: "entree", not "entrée".  don't you need those?

Fvi589, 2020-06-10:
It looks good. Yes I forced myself to put no accent, because I thought it was in HTML. But if the accents are interpreted, I would do a file with the accents, because in French, this is important and it helps understanding. I will review with the components if my translations are relevant, and I will adapt, because translating without the context, sometimes it is not easy

Fvi589, 2020-06-10:
for compilation, I will look. All IDEs are good, but as a former developer, too many options without going to the basics, which makes it turn into a gas plant (French expression)

Fvi589, 2020-06-13:
French version with accents. I did not know the synonymous expression for gas plant !!! on the other hand, really unfortunate that I can't compile the app, because I have lots of ideas for developing components. No one in the community can help me? because I know how to develop in Java.
[locale_fr - V1.txt](https://github.com/sharpie7/circuitjs1/files/4774966/locale_fr.-.V1.txt)

pfalstad, 2020-06-13:
Ok thanks, I have updated the web site.  http://www.falstad.com/circuit/circuitjs.html?lang=fr

Did you try updating the JRE in Eclipse?
[image]

Fvi589, 2020-06-14:
A priori, as we say in the IT jargon in France, "it fell on the march". I just managed to compile and launch the app. VICTORY !!! I will be able to start having fun !!! but in my opinion, it will take time before I understand all the workings of the app. But we finally managed to fold Eclipse !!! :)

ENJOY !!! :)

Fvi589, 2020-06-14:
I just managed to make a basic circuit breaker. It works. I created a library for "industrial components". On the other hand, I think that I would need help on the graphic library. Is there a link to see how to create graphic shapes? In any case, I like it and I have lots of ideas !!!
ENJOY ++ !!! :)

pfalstad, 2020-06-14:
I recommend you just look at the other elements to see how they draw.  Try to find a similar one.  The Graphics object is based on the java built-in class, https://docs.oracle.com/javase/7/docs/api/java/awt/Graphics.html  But not everything is supported.

Or you could just use a javascript context to draw, which is more powerful, and everything is supported: http://www.gwtproject.org/javadoc/latest/com/google/gwt/canvas/dom/client/Context2d.html  See ResistorElm.java for an example.

It will help to use the interpPoint functions, which are weird but useful.

interpPoint(a, b, 0) == a
interpPoint(a, b, 1) == b
interpPoint(a, b, .5) == halfway between a and b
interpPoint(a, b, c, .5)  makes c == halfway between a and b
interpPoint(a, b, c, d, .5, 10):
   c = halfway between a and b, but offset 10 pixels to the left.
   d = halfway between a and b, but offset 10 pixels to the right.

Fvi589, 2020-06-15:
This is what I do, I draw inspiration from other elements to create my components. As they say, I'm not going to reinvent the wire for cutting butter! :) (French expression). Could you please show me how you would create these 2 components in void draw (Graphics g). the red dots are my connection points.
[Graphic Components.pdf](https://github.com/sharpie7/circuitjs1/files/4780318/Graphic.Components.pdf)

pfalstad, 2020-06-16:
For the circuit breaker, copy from SwitchElm.java

A contactor is just a relay, isn't it?  I'm confused by that symbol.  It's just a box with dots?  That's how it's represented in a circuit schematic?

Fvi589, 2020-06-17:
I am making good progress on the circuit breaker on operation (rating (with standardized values), thermal, magnetic). I made a generic curve and the trigger time depends on the thermal setting. For the magnetic, it falls in 10 ms. On the other hand, on the graphic part, I struggle !!! :) but it does not scare me too much, the most important for me is the operation. I'll send it to you soon.

Fvi589, 2020-06-17:
Another little question, I know I'm exaggerating !!! I would like to do modbus TCP com. I got an open source in JAVA. Could you tell me how we could compile it with LushProject. I'm sending you the bookstore. Thank you again for all your help.

[EasyModbusJava.zip](https://github.com/sharpie7/circuitjs1/files/4793969/EasyModbusJava.zip)

Fvi589, 2020-06-18:
I managed to graph my circuit breaker open and closed. By cons I absolutely do not understand how we manage the click on the component, if you could help me. I looked at the switch library but now I don't see how we manage the click

pfalstad, 2020-06-18:
Malheureusement, it only works for elements that inherit from SwitchElm.  toggle() gets called when the mouse is pressed, and mouseUp() gets called when the mouse is released.  You don't care about mouseUp(), so you can just handle toggle() to reset the breaker.

Fvi589, 2020-06-20:
Not obvious, but that's it, the circuit breaker is operational, with its graphic part.
I'll send you, this new component.

So, Next !!! :)

Enjoy !!!

Fvi589, 2020-06-21:
I have one last problem with the circuit breaker. I did a basic assembly, a voltage source, a resistor, my circuit breaker. When I click on the circuit breaker, I open and close the circuit breaker well and the current in the circuit is correct. 0A when it is open, xxA when the circuit breaker is closed.
My problem is when I open the circuit breaker graphically from my code section. Visually the circuit breaker is open, but for mounting, the circuit breaker is always closed. I think it comes from the fact that since I did not click on the circuit breaker, the info is not sent to cirsim. How can I do it please? Otherwise everything else works.
PS: I allowed myself to respond to another message. I started to look, it should be much less difficult than the circuit breaker. I’m looking for something I can do.
In any case, by looking at all the Java classes, I can tell you that what you have developed is VERY powerful.

pfalstad, 2020-06-21:
I don't understand the question, sorry.  I suspect I would need to see the code to answer it anyway.

Fvi589, 2020-06-22:
I send you the modified files. Another small problem, when I copy / paste or export, my components are not taken into account.
I created a work timer from the Relay component, to do a test. I will make a specific component (Timed Relay for example), so as not to disturb the library.
[client.zip](https://github.com/sharpie7/circuitjs1/files/4814721/client.zip)

Fvi589, 2020-06-23:
with this file

[CircuitBreakerElm.zip](https://github.com/sharpie7/circuitjs1/files/4818171/CircuitBreakerElm.zip)

pfalstad, 2020-06-26:
You need to add tps_declenchement here to match the constructor:

	String dump() {
	    return super.dump() + " " + Calibre + " " + Thermique + " " + Magnetique  + " " + PdC + " " + tps_declenchement; 
	}

Fvi589, 2020-06-28:
I still have the same problem with the tripping of the circuit breaker (see simulation in pdf). Everything happens as if the circuit saw the circuit breaker still closed. I think it comes from the fact that we do not reclique on the circuit breaker. I need help finding a solution. Otherwise I go to my industrial bookstore. I made the contactors. Concerning the equipment, I created rails for the components and the chutes. I continue my bookstore.
[circuit breaker.pdf](https://github.com/sharpie7/circuitjs1/files/4842541/circuit.breaker.pdf)
[war.zip](https://github.com/sharpie7/circuitjs1/files/4842542/war.zip)
[Essai Disjoncteur.txt](https://github.com/sharpie7/circuitjs1/files/4842547/Essai.Disjoncteur.txt)
file:///C:/Users/minipc/Desktop/STS_ET/Paul/circuitjs1-master/war/circuitjs.html?cct=$+0+0.000005+1.0312258501325766+50+5+50%0Ag+544+448+544+496+0%0A370+544+384+544+448+1+0%0Ar+544+288+544+176+0+0.001%0AR+544+176+544+112+0+0+40+100+0+0+0.5%0A413+544+288+544+384+0+1+false+0.25+16+160+70000+1000000000+1+1000000000000%0A

pfalstad, 2020-06-28:
these changes should help

@@ -358,12 +358,22 @@
            sim.stampResistor(nodes[0], nodes[1], resistance);
        }
        //void setCurrent(int vs, double c) { current = -c; }
+
+       // undo some of the changes parent class SwitchElm made
+       int getVoltageSourceCount() { return 0; }
+       boolean getConnection(int n1, int n2) { return true; }
+       boolean isWire() { return false; }
+
+       boolean nonLinear() { return true; }
+
        void stamp() {
            sim.stampNonLinear(nodes[0]);
            sim.stampNonLinear(nodes[1]);
-               if (position == 0)
-                       sim.stampVoltageSource(nodes[0], nodes[1], voltSource, 0);
-               sim.stampResistor(nodes[0], nodes[1], resistance);
+
+       // we stamp in doStep not here
+       //      if (position == 0)
+       //              sim.stampVoltageSource(nodes[0], nodes[1], voltSource, 0);
+       //      sim.stampResistor(nodes[0], nodes[1], resistance);
                /*
                double v = (position == 0) ? loV : hiV;
            if (isTernary())

pfalstad, 2020-06-28:
github mangled the diff I posted so let's try that again.  These changes should help.
 
[cir.zip](https://github.com/sharpie7/circuitjs1/files/4842866/cir.zip)

Inheriting from SwitchElm caused some problems.

Fvi589, 2020-06-29:
FANTASTIC , it works !!! :)
This solves a lot of problems for me, compared to the components I wanted to create. I will try in the week a COM modbus TCP, in order to be able to attack my models under Unity3D.

I think I have found the right tools for my lessons and I am very motivated.

do you think that the industrial bookstore could interest people? Of course, I make available to the community all the components that I will create.

ENJOY !!!

Fvi589, 2020-07-07:
I am making good progress in my library of industrial components. But I need the COM modbus TCP so that I can use with my students, all the great possibilities of the software. I tried it out, but I can't compile because it can't find the Java classes of the easymodbus opensource module. Sorry, i'm not a Java expert. 
If I compile, I would do the rest
Thanks

[src.zip](https://github.com/sharpie7/circuitjs1/files/4886089/src.zip)
[Paul-FVI [Modbus tcp].pdf](https://github.com/sharpie7/circuitjs1/files/4886090/Paul-FVI.Modbus.tcp.pdf)