sharpie7/circuitjs1 #447: Calculating methods

popkingcorn, 2020-05-28:
Hi,
What method/algorithm do you use to calculate the circuits?
Maybe you can recommend some literature to find out.

pfalstad, 2020-05-28:
This is the book I used, it's very good: https://www.amazon.com/Electronic-Circuit-System-Simulation-Methods/dp/0070501696

colinhowell, 2020-05-28:
There's also the original papers on SPICE:

https://www2.eecs.berkeley.edu/Pubs/TechRpts/1975/9602.html
https://www2.eecs.berkeley.edu/Pubs/TechRpts/1973/22871.html

They're old, but they provide a useful explanation of how this stuff is done. The first-listed paper is later (1975) but is much longer and goes into far more detail; I also included the second-listed paper (from 1973) because it adds more background on SPICE's development.

SPICE is an ancient program, but the methods still work--and compared to the original low-end supercomputers they first ran on, modern hardware is fast enough to run them on the fly in interactive Javascript!

popkingcorn, 2020-05-28:
So circuit calculated by solving set of nodal equations based on KCL (lu_solve function in code), did I understand correctly?

pfalstad, 2020-05-29:
That’s right.

popkingcorn, 2020-05-29:
Thanks for answers!

closed 2020-05-29 by popkingcorn