sharpie7/circuitjs1 #451: Feature request / Timer

netpsa, 2020-06-18:
Hi. First of all thanks by the tool.

I just missing one simple timer for logic aplications both "timer on" and "timer off".

br

pfalstad, 2020-06-18:
Is there a particular device you have in mind?  Do you have a link to a datasheet?

netpsa, 2020-06-19:
Maybe its already implemented and I don't know how to use. I think on a simple block to perform delay to turn on or turn off only to build logic diagrams. Timer ON (Xsec)  input = 1, output = 1 X seconds laterTimer OFF (Xsec)  input = 1, output = 0 X seconds laterImage attached. Thank you for the attention. Enviado do meu smartphone Samsung Galaxy.
-------- Mensagem original --------De : Paul Falstad <notifications@github.com> Data: 18/06/2020  14:05  (GMT-03:00) Para: sharpie7/circuitjs1 <circuitjs1@noreply.github.com> Cc: netpsa <netpsa@uol.com.br>, Author <author@noreply.github.com> Assunto: Re: [sharpie7/circuitjs1] Feature request / Timer (#451) 
Is there a particular device you have in mind?  Do you have a link to a datasheet?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-646178748",
"url": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-646178748",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Fvi589, 2020-06-20:
Hi,
this kind of timer ?
http://blog.formatis.pro/rt-relais-temporises

netpsa, 2020-06-20:
Hi, Yes, this kind of timer. However more simple yet for only 2 functions:
-------- Mensagem original --------De : Fvi589 <notifications@github.com> Data: 20/06/2020  06:37  (GMT-03:00) Para: sharpie7/circuitjs1 <circuitjs1@noreply.github.com> Cc: netpsa <netpsa@uol.com.br>, Author <author@noreply.github.com> Assunto: Re: [sharpie7/circuitjs1] Feature request / Timer (#451) 
Hi,
this kind of timer ?
http://blog.formatis.pro/rt-relais-temporises

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-646969296",
"url": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-646969296",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Fvi589, 2020-06-21:
I will try to make you a work and rest tempo from the RELAY component during the week. Please be indulgent, I do not master the JAVA development of the app. :) English either, thank you google translation

netpsa, 2020-06-22:
Hi, Would be very nice if you can include this feature. Also english and java isn't that easy to me :-)... I suguest to make the timer very simple, no need to a advanced one... Att.
-------- Mensagem original --------De : Fvi589 <notifications@github.com> Data: 21/06/2020  13:47  (GMT-03:00) Para: sharpie7/circuitjs1 <circuitjs1@noreply.github.com> Cc: netpsa <netpsa@uol.com.br>, Author <author@noreply.github.com> Assunto: Re: [sharpie7/circuitjs1] Feature request / Timer (#451) 
I will try to make you a work and rest tempo from the RELAY component during the week. Please be indulgent, I do not master the JAVA development of the app. :) English either, thank you google translation

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-647152851",
"url": "https://github.com/sharpie7/circuitjs1/issues/451#issuecomment-647152851",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Fvi589, 2020-06-22:
I made a work timer from the Relay component. I would do something cleaner, it's just for testing. In Edit, you can set the tempo time. BE CAREFUL, if you put 5sec, you have to think about changing the time base which is 5µs by default.
Launch :
War -> circuitjs.html
[image]
[war.zip](https://github.com/sharpie7/circuitjs1/files/4814657/war.zip)
ENJOY !!! :)

dumoulinalex, 2020-10-30:
The provided modified relay component is promising, but unfortunately doesn't work fully. The changing branch is energized after the timer as expected, but it cannot be reset and the other branch (NC) is not "connected" electrically at the start of the simulation.

I also have a strong interest in a timer component. In the power industry, temporised relay such as the ones described by Fvi589's link (June 20 post) are used for coordinating protection function. Such timers are used in conjuction with logic gates and contacts to create control schemes.

The behavior of temporised relay is as such:
[image]

For instance, one might want to close a specific breaker but not until 1s has passed since another breaker was opened; the 'breaker A is open' contact might be wired at the activating coil of a timer relay with a pick-up setting of 1s, with its output contact closing breaker B.

Another example using the drop-out feature is when you activate a trip coil, you might want to keep the trip coil energized for a certain amount of time (say, 1s), to be sure it trips. First, a protection device would momentarily energize the coil of the timer relay (adjusted with a drop-out setting of 1s). This signal is usually only a short pulse, but the drop-out feature of the timer relay keeps its output activated for the set amount of time (1 second in this case) before returning to a de-energized state.

Modern protection relays are fully digital and use FPGA functions to emulate the conventional behavior of those temporized relay. I used the Falstad circuit simulator extensively while in school and would love to implement my protection logic in it as well, but I am hindered by the lack of a timer function. (I made a semi-functional hack using transmission lines as a delay)

pfalstad, 2020-12-11:
Added time delay relay

closed 2020-12-11 by pfalstad