sharpie7/circuitjs1 #453: Current Limiter

skoehler, 2020-06-21:
The drivers inside logic ICs typically are current limited. What I would like to do is to simulate an open drain output of a logic IC that pulls a bus line to ground with a constant current, say 5mA.

I have tried a current source and a switch. That has the desired effect, until the line has a negative voltage. That is not intended.

Please add some sort of device that limits the current to a specific amount regardless of the voltage.

[image]
[circuit-20200621-1157.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4809278/circuit-20200621-1157.circuitjs.txt)

skoehler, 2020-06-21:
I found out, that I can set the slew rate of the output of a logic inverter.
Now when I watch the output of the logic inverter in a scope, it does not behave as expected. The color of the output slowly fades between green and grey, but in the scope I see a sharp edge.
When I connect the output of the logic analyzer the above circuit, it also doesn't behave as expected. Assume that the output is low (0V). When I close the switch, I would expect the voltage of the capacitor to slowly drop. But it doesn't. It jumps to 0 immediately.

I also replaced the current mirror with a resistor. But the effect is the same.

skoehler, 2020-06-21:
I don't think that setting the slew rate directly is a good thing. I'm not an electrical engineer, but I believe that the slew rate is merely a result of the driver's current and the capacitance of the data lines. I am pretty sure that you are trying to approximate the behavior of a real-world logical circuit, but I don't think that this fits together with the rest of your simulator (which simulates analog circuits). If the current is not defined, I should not be able to connect a logical output to an analog circuit.

pfalstad, 2020-06-21:
A current limiter would be a good addition.  Also a real current source.  The current source in the app is ideal and will go on forever no matter what the voltage is, which isn't realistic.

You can use a real current source to do the thing you want, like so: http://tinyurl.com/y9268g4j

The inverter slew rate seems to work for me.  Do you have an example circuit?  I think that was added to handle the inverter oscillator example, where having no-delay inverters would not work.

skoehler, 2020-06-22:
Thank you for considering a current limiter as an addition.

I do not understand the current source that you posted, but it works as desired.

Regarding the slew rate, find an example attached. Close the switch. I see a sharp edge, new slew at all.
[circuit-20200622-1350.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4813006/circuit-20200622-1350.circuitjs.txt)

pfalstad, 2020-06-22:
Yeah that's a bug.  If you use it in an analog circuit where it takes multiple iterations, then the inverter output will move faster than the slew rate.

Here is a (brief) explanation of the current source: http://www.falstad.com/circuit/e-currentsrc.html