sharpie7/circuitjs1 #454: Problem with plotting Lissajous images

Roger-Hlls, 2020-06-24:
Hi all!

   I just started using Electron and I tried to make some Lissajous images. Among four (4) ".txt" files I made and saved, two (2) of them when are re-opened don't display correctly their respective plots... The curious thing is that I made each from a starting circuit and I only changed the frequency of one (1) circuit (of two -2) from file to file before saving anyone. Any ideas?

Roger-Hlls
[circuit-20200624_2_2_4_a2.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4823452/circuit-20200624_2_2_4_a2.circuitjs.txt)
[circuit-20200624_2_2_4_b2.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4823453/circuit-20200624_2_2_4_b2.circuitjs.txt)
[circuit-20200624_2_2_4_c2.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4823454/circuit-20200624_2_2_4_c2.circuitjs.txt)
[circuit-20200624_2_2_4_d2.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4823455/circuit-20200624_2_2_4_d2.circuitjs.txt)

pfalstad, 2020-06-24:
Hi, which ones are incorrect and what's incorrect about them?

Roger-Hlls, 2020-06-24:
Hi!

 

                Thank you for contacting me. The correct files are the:

 

circuit-20200624_2_2_4_b2.circuitjs.txt

circuit-20200624_2_2_4_c2.circuitjs.txt

 

where the Lissajous plots are re-opened as they should (roughly an “8”, laid on each axis respectively). Thank you for the immediate attention,

 

Roger-Hlls

pfalstad, 2020-06-24:
Ok I'm guessing that before you saved these, like "a" for example, there was a phase difference that made it draw a circle instead of a straight line.  This phase difference can happen if you play around with the frequency of a voltage source.  That causes the sim to silently add on a phase difference to make sure the waveform doesn't have a discontinuity.  But this phase difference gets lost if the circuit is saved or if you press "Reset".  If you want a permanent phase offset, you have to fill in the "Phase Offset" field when editing the voltage source.  And then press "Reset" to make sure it looks like you want it to look before saving.

Roger-Hlls, 2020-06-24:
Well,

 

                It may be so, however I first drew / saved the wrong file “circuit-20200624_2_2_4_b2.circuitjs.txt”, although the plot seemed correct, then I changed the frequency of the first circuit, saw the correct plot and saved the file, and so on up to the last file. Next, I went on doing other simulations; yet, opening the files again in  a future time, I happened to the problem... I never filled any “Phase Offset”, because I don’t need to set up one. I’ll try your suggestion, though, still I’d like a more suitable solution. Thank you,

 

Roger-Hlls

closed 2020-12-05 by sharpie7