sharpie7/circuitjs1 #455: Problem using subcircuit

bwest81, 2020-06-28:
I created a subcircuit without any issues but when I try to create an instance of it, nothing happens.  I know I am doing everything correctly because if I create a simple circuit and do the same, it works fine.  If I add an instance of another subcircuit and then try to switch it to my problem circuit, it doesnt work and I can no longer add anymore subcircuits.  Attached is the circuit in question

[circuit-20200627-2113.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4841689/circuit-20200627-2113.circuitjs.txt)

Thank you

pfalstad, 2020-06-28:
Yes, seems to be the same as #442.  I'll fix it to give an error instead.

sharpie7, 2020-08-02:
Fixed

closed 2020-08-02 by sharpie7