sharpie7/circuitjs1 #457: Op-Amps: Problem with non-inverting / follower amplifier

Roger-Hlls, 2020-07-03:
Hi all!

   I've made the "e1_3_3.txt" circuit as a non-inverting op-amp amplifier and the "e1_3_4.txt" one as an op-amp follower. I based my circuits on the LM741 model; however, the y don't seem to work correctly, not in a real-life circuit on the Lab with the same components. More specifically, the output has a 180o phase distortion! I also tried othe op-amp models with the same results. Any ideas?

Roger-Hlls
[e1_3_3.txt](https://github.com/sharpie7/circuitjs1/files/4869189/e1_3_3.txt)
[e1_3_4.txt](https://github.com/sharpie7/circuitjs1/files/4869190/e1_3_4.txt)

pfalstad, 2020-07-03:
I think it's working, it's just that the input source is upside down.  The * side is the positive side, and should be on top.  If you look closely at color of the wires you can see that the input to the op-amp and the output are in phase.  But the scope is showing it reversed because the polarity is reversed.

Roger-Hlls, 2020-07-03:
Thank you very much Paul!

 

                However, since I’m using the MS-Windows version, I tell you I cannot see the “*” my friend... Is there a bug? On the other hand, you took a lot of stress out of me! Many thanks again,

 

Roger-Hlls

pfalstad, 2020-07-03:
I need to update the windows version with the * change.

Roger-Hlls, 2020-07-03:
Dear Paul,

 

                Wonderful! Thank you again,

 

Roger-Hlls

 

P.S.        I could send you my oscillator images of real-world “Electronics 101” circuits and the output of the simulator ones they don’t agree (e.g. output of a NAND ring oscillator). Are you interested?

closed 2020-12-05 by sharpie7