sharpie7/circuitjs1 #458: Ring Oscillators

Roger-Hlls, 2020-07-04:
Hi all!

   I built a ring oscillator by NAND 74xx00 gates, both TTL and CMOS material; i.e. a chain of 7, at least, 74LS00 or 7aHC00 NAND gates with short-circuited inputs, forwarded their outputs to the next gate in tow. The last NAND gate, also the output of the ring, feed-backed its output to the (short-circuited together) inputs of the first one. I supply the circuit for 'electron':

[e2_3_1_a.txt](https://github.com/sharpie7/circuitjs1/files/4872400/e2_3_1_a.txt)

However, in my Lab, by using real-world gate I obtained the results shown in files:

[image]
[image]
for the 'LS' chip and files:

[image]
[image]
for the HC ones.

I used a Tektronix TDS 2002B oscilloscope; '1a_1' / '1b_1' files show 50.0nsec time divisions, whereas '1a_2' / '1b_2' ones 25.0nsec time divisions. So, is there any chance to obtain better results in 'electron'? Also, a not so 'thick' waveform?

   Thank you,

Roger-Hlls

pfalstad, 2020-07-04:
The NAND gates in the simulator are ideal, responding instantly (in one time step) and with an infinite slew rate.  If you’re going to build a ring oscillator like this, the behavior will depend on the real response of whatever gate implementation you are using.  So you can’t use ideal gates to model it.  You’ll have to get the schematic for the NAND gates and use it to make a sub circuit and then use that to make the ring oscillator.

Roger-Hlls, 2020-07-05:
Hi Paul,

 

                Yep, knew it, yet I guess I hoped for another solution... Also, you didn’t give me any insight on the ‘density’ of the plot and how to further ‘spread’ it to make it more clear... Thank you for your answer,

 

Roger-Hlls

pfalstad, 2020-07-05:
You want to zoom in on the scope more so you can see the waveform better?  Unfortunately you've zoomed in as far as possible.  Normally I'd say that you need to make the timestep smaller, but this broken circuit will go as fast as the timestep will allow, since there's no capacitors or anything.  If you set the timestep small enough, it will produce gamma rays.

Roger-Hlls, 2020-07-05:
Thanks, Paul, a very good, humorous, answer! You’ re a pal,

 

Roger-Hlls

closed 2020-09-10 by pfalstad