sharpie7/circuitjs1 #460: Problems with transistor circuits

Roger-Hlls, 2020-07-08:
Hi all!

	I tried to run some transistor-based circuits by means of the simulator and I encountered some problems. Any help?

Emitter follower
The result I got, as you can see, is not as expected... However the structure of the circuit (i.e. the ratios of resistors and its building blocks), I think, is alike to that given as a test bed, the file:
	Circuits >>> Transistors >>> Emitter Follower
[e4_2_4_a.txt](https://github.com/sharpie7/circuitjs1/files/4889444/e4_2_4_a.txt)

Common emitter amplifier
The result I got, as you can see, is a semi-rectified output. However the structure of the circuit (i.e. the ratios of resistors and its building blocks), I think, is the same as that given as a test bed, the file:
	Circuits >>> Transistors >>> Common-Emitter Amplifier
[e5_2_1_b.txt](https://github.com/sharpie7/circuitjs1/files/4889451/e5_2_1_b.txt)

This time I'm sure the generator poles are properly set. Moreover, I tried a number of different circuits from the literature and I got the same results. Any ideas? Thanks in advance,

Roger-Hlls

P.S.	I hope the answer wouldn't lie on the real-world vs. simulated-one difference

pfalstad, 2020-07-08:
I think in both cases the transistor isn't biased properly.  In the first case, the base swings as low as -15 V, but the emitter cannot go below ground, so the waveform is getting clipped.

Roger-Hlls, 2020-07-08:
Hi Paul,

 

                Well, as I said, the circuits ‘work’ fine with real-world materials, I can send you my oscilloscope pictures. Also, especially in the second case, the resistor ratios and the bias volts are properly set. Would you like me to send you the circuits found in the literature? Thank you,

 

Roger-Hlls

pfalstad, 2020-07-08:
Sure, send me where you found them, and your scope pictures, for the first circuit at least.

Roger-Hlls, 2020-07-08:
Hi again!

 

                You know, I fell very obliged to you for your simulator and the prompt responses... On the job; I mostly read the “Principles and Applications of Electrical Engineering” by Giorgio Rizzoni, and “Microelectronic Circuits” be Sedra - Smith. Their analyses led me to build the circuits I sent you... Now, for my oscilloscope image:

 

Common emitter amplifier

 

I resolved the issue with the emitter follower one, a faulty transistor was the cause. Thank you always,

 

Roger-Hlls

colinhowell, 2020-07-08:
None of these pictures have been getting through to the bug report.

colinhowell, 2020-07-08:
The problem with the first circuit, the emitter follower, seems to have been that, given the high frequency (10 kHz) of the AC source, the coupling capacitor was large enough to let the full +/- 15 V AC through from the source to the transistor base. So the transistor couldn't possibly avoid being driven into cutoff during the negative half of each cycle.

In the original Emitter Follower example, while the coupling capacitor is of comparable magnitude, the source frequency is much lower, and so are the biasing resistors. Together, the coupling capacitor and biasing resistors form a high-pass filter which attenuates the amplitude of the AC source enough to keep the base voltage always positive, so the transistor is never cut off.

To achieve this for the given circuit, compensating for its higher frequency and larger resistances, the coupling capacitor must be reduced from 1 uF down to around 100 pF. The amplitude of the output signal will be lower, just as that of the original Emitter Follower is relative to the source, but the signal will not be clipped or cut off.

pfalstad, 2020-07-08:
As Colin pointed out, I don't see any pictures.

For the amplifier, the transistor base is biased to 1.355 V with no input, so I don't understand how you're managing to amplify a 2 V signal without some clipping.

Roger-Hlls, 2020-07-09:
Hi Colin,

 

                Thank you, I’ll provide them by another procedure,

 

Roger-Hlls

Roger-Hlls, 2020-07-09:
Hi Colin (again),

 

                You’ re right, The performance is what you describe even in my lab. I used a faulty transistor, that is why I mistook by asking help for the emitter follower circuit. I already told Paul, so. On the other hand, there remains the problem with the common emitter amplifier circuit; for that I’ll provide an image with another procedure in my next message, in order to answer to Paul’s one. Thank you,

 

Roger-Hlls

Roger-Hlls, 2020-07-09:
Hi Paul,

 

                A funny thing... I couldn’t find my ‘issue’ to add the image... Anyhow, here it is, as an attachment, for my common emitter amplifier, I hope it’ll pass through. Thank you,

 

Roger-Hlls

pfalstad, 2020-07-09:
Still don't see an attachment.

pfalstad, 2020-07-09:
Go to github using the link and post it there, don't email it.

Roger-Hlls, 2020-07-09:
Well,

 

                Certainly I’m not sending a virus! So, a final try, should this fail too, I’ll start a new issue. Just rename the “.f” file to “.jpg”. Thank you,

 

Roger-Hlls

Roger-Hlls, 2020-07-09:
Paul,

 

                Told you, can’t find my original issue... However, should my penultimate message to you not ‘carry’ the image, I’ll start a new issue to embed it. Thanks,

 

Roger-Hlls

pfalstad, 2020-07-09:
Here is a link to it: https://github.com/sharpie7/circuitjs1/issues/460#issuecomment-656004457

Roger-Hlls, 2020-07-09:
Thanks Paul!

   So, the (in)famous image...

[image]

Roger-Hlls

pfalstad, 2020-07-09:
Great.  So what am I looking at here..  I assume CH1 is the first probe, of the emitter voltage, and CH2 is the output?  How do I read the scale..  What are the high and low voltage values of CH1 and CH2?  And do you have a scope view of the transistor base voltage?

Roger-Hlls, 2020-07-09:
Hi pal!

	All legitimate questions! So, let me see...

-	CH1 is the first probe, the input; it’s a sine waveform with data <2.00Vpp, @20.00KHz>
-	CH2 is the output and it comes from the <680.00Ω> resistor when the RC filter (<1.00KΩ> and <C = 4.70μFd>) is attached to the transistor’s emitter
-	The scale is both 1.00V / DIV
-	High voltages for both channels are -1.00V / +1.00V
-	I made the scope you asked, it's in the attached file
[image]

So, let's hope we'll find the solution this time... Thank you,

Roger-Hlls

colinhowell, 2020-07-09:
So, looking at your second circuit and your discussion of it, I'm afraid that a couple of things seem confused here.

You identify the input as 2.00 Vpp (2 volts peak-to-peak), both in your model and when discussing your oscilloscope trace. The oscilloscope trace is consistent with that, since it's set to 1 V/div and the input waveform spans 2 divisions vertically, so it swings from plus 1 volt to minus 1 volt, a 2-volt span. But in your circuit model, the input is set to Max Voltage: 2, which means it will swing from plus 2 volts to minus 2 volts. That's 4 volts peak-to-peak, not 2. "Max Voltage" sets the AC amplitude, not its peak-to-peak range.

More seriously, you're describing this circuit as a common-emitter amplifier, and comparing it to a common-emitter example, but your model still uses the emitter as its output, just like in the emitter follower circuit. Furthermore, you describe your lab circuit in the same way, saying that your oscilloscope trace of the output was taken from the emitter resistor. But a common-emitter amplifier uses the transistor's *collector* terminal as the output, and the oscilloscope trace you show of the output (roughly 2 volts peak-to-peak, centered on 7.13 volts) is consistent with what you would expect to see on the collector, not the emitter. I don't see how you could get that trace if it were really taken from the transistor's emitter terminal.

I altered your model to have a true 2 V peak-to-peak input (Max Voltage: 1) and to have the output connected to the transistor collector instead of the emitter. (I also swapped the terminals on the polarized output cap, since as it was originally placed, my changes eventually would make it complain about excessive reverse voltage.) Once I ran the model long enough for it to adjust to the changes, it started to look much like your oscilloscope traces. The output voltage was centered at about 7.1 volts, with a range of slightly over 2 volts peak-to-peak, and *flattened tops* (look closely at your output oscilloscope trace). The base voltage had a similar range to that on the oscilloscope: both stay above ground throughout but swing quite near ground at the low points. That low base voltage is related to the output's flattened tops: your input is getting too close to the limits of the available voltage range, which distorts the output.

pfalstad, 2020-07-09:
I think Colin has it (thanks Colin!) but I still can't read the scale.  You said "High voltages for both channels are -1.00V / +1.00V" (I assume you mean high/low voltages) but why is one drawn 3V above the other?  Where is ground, is it in the middle?  The third scope that I asked for, does that have the same scale?  It says 500 mV in the lower left, is that the volts/div?

colinhowell, 2020-07-09:
I actually looked up the oscilloscope manual to figure all that out. In his earlier issue #458, he said the oscilloscope was a Tektronix TDS 2002B, whose manual is [here](https://www.tek.com/oscilloscope/tds1001b-manual/tds1000b-and-tds2000b-series). Check pages 32-34 of the PDF, which are labeled as pages 10-12 of the printed manual: those identify the various symbols and readouts on the screen, including numbered brief descriptions that I quote below.

If I understand right, the little colored arrow markers on the left margin show ground for each channel (quote: "6. On-screen markers show the ground reference points of the displayed waveforms."). So in the first screenshot, CH1's ground level is in the center of its waveform, while CH2's is down below the bottom, and that waveform has been shifted to show clearly on the screen. In the second screenshot, ground is a little below the bottom of the waveform.

Vertical scale of the channels is shown at the bottom left (quote: "8. Readouts show the vertical scale factors of the channels."). So the first screenshot is 1.00 V/div for both waveforms, while the second is 500 mV/div. Timescale is next to those readouts: 25 µs/div.

Then there are the little markers at the right edge of the scope, and the corresponding readouts at the lower right. For the markers, quote: "5. Marker shows Edge or Pulse Width trigger level." For the readouts, quote: "14. Readout shows Edge or Pulse Width trigger level." Here the scope is in Edge trigger mode, triggered by the rising edge (see the little icon before the readout). The position of the trigger is shown by the black arrow marker at the scope's top edge; here it's always in the center of the screen. The trigger level markers can then be used as voltage level indicators; note they match the channel's signal level at the trigger position. The trigger level in the first screenshot is at 7.13 V, which applies to CH2 only. (That explains why CH2's ground level is below the scope's bottom edge.) The trigger level in the second screenshot is 1.12 V, consistent with the signal's ground level being a little more than 2 divisions further down at 500 mV/div, and with the whole waveform being 2.00V peak-to-peak.

Anyway, that's how I figured out the state of the circuit in my earlier report.

Roger-Hlls, 2020-07-09:
Hi Colin,

 

                Happy to answer all questions, very happy to see so much willingness to help me! I feel, again, very obliged...

 

> From: Colin Douglas Howell [mailto:notifications@github.com] 
> Sent: Thursday, July 09, 2020 4:52 PM
> To: sharpie7/circuitjs1
> Cc: Roger-Hlls; Author
> Subject: Re: [sharpie7/circuitjs1] Problems with transistor circuits (#460)

> 

> So, looking at your second circuit and your discussion of it, I'm afraid that a couple of things seem confused here

> 

> You identify the input as 2.00 Vpp (2 volts peak-to-peak), both in your model and when discussing your oscilloscope trace. The oscilloscope trace is consistent with that, since it's set to 1 V/div and the input waveform spans 2 divisions vertically, so it swings from plus 1 volt to minus 1 volt,

> a 2-volt span. But in your circuit model, the input is set to Max Voltage: 2, which means it will swing from plus 2 volts to minus 2 volts. That's 4 volts peak-to-peak, not 2. "Max Voltage" sets the AC amplitude, not its peak-to-peak range

I didn’t know that, I thought ‘max voltage’ meant ‘peak-to-peak’. Good to know, thank you, I’ll change that in all my (simulation) files

 

> More seriously, you're describing this circuit as a common-emitter amplifier, and comparing it to a common-emitter example, but your model still uses the emitter as its output, just like in the emitter follower circuit. Furthermore, you describe your lab circuit in the same way, saying that your

> oscilloscope trace of the output was taken from the emitter resistor. But a common-emitter amplifier uses the transistor's collector terminal as the output, and the oscilloscope trace you show of the output (roughly 2 volts peak-to-peak, centered on 7.13 volts) is consistent with what you

> would expect to see on the collector, not the emitter. I don't see how you could get that trace if it were really taken from the transistor's emitter terminal

Hmm... OK, I took the circuits from a Professor’s notes, he used to refer to the circuits that way. However, I see your point and it is valid. On the other hand, all my plots are correct, taken from the node I specified

 

> I altered your model to have a true 2 V peak-to-peak input (Max Voltage: 1) and to have the output connected to the transistor collector instead of the emitter. (I also swapped the terminals on the polarized output cap, since as it was originally placed, my changes eventually would make it

> complain about excessive reverse voltage.)

Hmm... I’m OK with the first changes, however I do not agree with the swapping; you see, Paul told me about the ‘polarity’ of the generator and I agreed to the swapping. To solve the excessive reverse voltage problem, I simply ‘tweaked’ the related value of the capacitor

 

> Once I ran the model long enough for it to adjust to the changes, it started to look much like your oscilloscope traces

Hoorah! It’s always good to know you’re not an elephant (I saying in my country; being an elephant means not being correct in your assessment)

 

> The output voltage was centered at about 7.1 volts, with a range of slightly over 2 volts peak-to-peak, and flattened tops (look closely at your output oscilloscope trace)

Correct

 

> The base voltage had a similar range to that on the oscilloscope: both stay above ground throughout but swing quite near ground at the low points. That low base voltage is related to the output's flattened tops: your input is getting too close to the limits of the available voltage range, which

> distorts the output

Correct, again

 

                Thanks,

 

Roger-Hlls

Roger-Hlls, 2020-07-09:
Hi Paul,

 

                Some answers:

 

> From: Paul Falstad [mailto:notifications@github.com] 
> Sent: Thursday, July 09, 2020 10:49 PM
> To: sharpie7/circuitjs1
> Cc: Roger-Hlls; Author
> Subject: Re: [sharpie7/circuitjs1] Problems with transistor circuits (#460)

> 

> I think Colin has it (thanks Colin!) but I still can't read the scale. You said "High voltages for both channels are -1.00V / +1.00V" (I assume you mean high/low voltages) but why is one drawn 3V above the other?

I simply set the channel trace in a cosy ‘altitude’, I didn’t mean to mark the ‘ground’ line. Also, since the scale is 500mV/division, that means that ~3.80divisions x 500mV = ~1.90V(peak-to-peak)

 

> Where is ground, is it in the middle?

In the middle of the waveform (sine), yes

 

> The third scope that I asked for, does that have the same scale?

No, it’s 500mV/division, I tried give a more ‘zoomed out’ image

 

> It says 500 mV in the lower left, is that the volts/div?

Yes

 

                Thanks,

 

Roger-Hlls

Roger-Hlls, 2020-07-09:
Hi Colin,

 

                Two messages in a row! You over-indulge me... :- )) All you quote are correct, you put on a tremendous effort! Thanks,

 

Roger-Hlls

pfalstad, 2020-07-09:
Nice work Colin.  I did some google image searches on tek scopes and then gave up.

Are you really using polarized capacitors?  This doesn't seem like a good application for them.  I would use regular capacitors.

Roger-Hlls, 2020-07-10:
Hi Paul,

Hi Colin,

 

                Wasn’t sure whether you referred to me but, yes, in my real-world circuits I also use polarized capacitors. Using ‘electron’, though, that doesn’t seem necessary. Thanks,

 

Roger-Hlls

colinhowell, 2020-07-10:
> Hi Colin, Happy to answer all questions, very happy to see so much willingness to help me! I feel, again, very obliged...

>> From: Colin Douglas Howell ***@***.*** Sent: Thursday, July 09, 2020 4:52 PM To: sharpie7/circuitjs1 Cc: Roger-Hlls; Author Subject: Re: [sharpie7/circuitjs1] Problems with transistor circuits (#460)

[snip]

> I altered your model to have a true 2 V peak-to-peak input (Max Voltage: 1) and to have the output connected to the transistor collector instead of the emitter. (I also swapped the terminals on the [...]

> Hmm... I’m OK with the first changes, however I do not agree with the swapping; you see, Paul told me about the ‘polarity’ of the generator and I agreed to the swapping. To solve the excessive reverse voltage problem, I simply ‘tweaked’ the related value of the capacitor

Oh, in this case swapping the capacitor terminals was just my way to get the simulation not to complain about that. Tweaking the reverse voltage is another (in fact at one point I was doing that instead). I suppose the better alternative would have been to use a nonpolarized capacitor instead.

The real issue, as Paul has already pointed out, is why are you using polarized capacitors there in the first place? Did you just happen to pick them in the sim, or are you using them in the real circuit for some reason? Seems a bad idea to use such capacitors for an obviously oscillatory signal unless you have no choice, in which case you must *guarantee* that the voltages will always go the proper way. Polarized capacitors only tolerate being charged in one direction, with the + terminal being at more positive voltage than the - terminal. They will actually be permanently damaged or destroyed (sometimes even exploding!) if those voltages are reversed. That's the real reason for the simulator warning, as well as why I chose to swap the capacitor terminals—I didn't want to set a bad example. If you're going to make something work with a hack, try to make it a safe hack.

Roger-Hlls, 2020-07-10:
Good morning my friend (at least at EET +02:00),

 

                I am stunned these days to ever realize how much you (Paul and you) love your work and how much all of us, Paul, Colin, and me, would like to support an excellent job (i.e. the simulator)! I’d also like to tell you I plan to widespread it to my department and make it the #1 electronics simulation tool! Now, on to your message:

 

> From: Colin Douglas Howell [mailto:notifications@github.com] 
> Sent: Friday, July 10, 2020 9:34 AM
> To: sharpie7/circuitjs1
> Cc: Roger-Hlls; Author
> Subject: Re: [sharpie7/circuitjs1] Problems with transistor circuits (#460)

> 

>> Hi Colin, Happy to answer all questions, very happy to see so much willingness to help me! I feel, again, very obliged...

 

> From: Colin Douglas Howell @.*** Sent: Thursday, July 09, 2020 4:52 PM To: sharpie7/circuitjs1 Cc: Roger-Hlls; Author Subject: Re: [sharpie7/circuitjs1] Problems with transistor circuits (#460 <https://github.com/sharpie7/circuitjs1/issues/460> )

> [snip]

 

>>> I altered your model to have a true 2 V peak-to-peak input (Max Voltage: 1) and to have the output connected to the transistor collector instead of the emitter. (I also swapped the terminals on the polarized output cap, since as it was originally placed, my changes eventually would make

>>> it complain about excessive reverse voltage.)

>> Hmm... I’m OK with the first changes, however I do not agree with the swapping; you see, Paul told me about the ‘polarity’ of the generator and I agreed to the swapping. To solve the excessive reverse voltage problem, I simply ‘tweaked’ the related value of the capacitor

 

> Oh, in this case swapping the capacitor terminals was just my way to get the simulation not to complain about that. Tweaking the reverse voltage is another (in fact at one point I was doing that instead). I suppose the better alternative would have been to use a nonpolarized capacitor

> instead

Yes, for using the ‘electron’ per se  purposes

 

> The real issue, as Paul has already pointed out, is why are you using polarized capacitors there in the first place? Did you just happen to pick them in the sim, or are you using them in the real circuit for some reason?

I have to run an electronics Lab and to teach my students the value of all components... On the other hand, I have specifically inductors! :- ))

 

> Seems a bad idea to use such capacitors for an obviously oscillatory signal unless you have no choice, in which case you must guarantee that the voltages will always go the proper way. Polarized capacitors only tolerate being charged in one direction, with the + terminal being at more

> positive voltage than the - terminal. They will actually be permanently damaged or destroyed (sometimes even exploding!) if those voltages are reversed

How true... On the other hand, polarized capacitors can also explode by higher frequencies than those they’re designed for! It’s in their impedance formula [Zc = 1 / (jωC), ω = 2 * π * f, and C: capacitance value]

 

> That's the real reason for the simulator warning, as well as why I chose to swap the capacitor terminals—I didn't want to set a bad example. If you're going to make something work with a hack, try to make it a safe hack

Also, how true...

 

                Thanks for the ‘chat’, Colin, see you in my next question!

 

Roger-Hlls

pfalstad, 2020-12-11:
Don't believe there's a bug here, correct me if I'm wrong.

closed 2020-12-11 by pfalstad