sharpie7/circuitjs1 #461: Request: Simulated data download

Sakastian, 2020-07-08:
Hi:

In these times of pandemic, where it is difficult to go to a laboratory, it would be great to be able to download the data from the oscilloscopes. Something like: Right-click on the oscilloscope and then select "download data". The idea is that the file is similar to those obtained from a real oscilloscope. A .txt file with CSV would be great. In this way, you could make charts in Excel, for example, to later perform calculations, superimpose images, highlight parts of the waveform, use the info for a "lab" report, etc.
Very good work with the simulator! Congratulations.
Greetings from Chile.

Sebastián Arroyo Klein

pfalstad, 2020-07-08:
Hi, under "Outputs and Labels" there is an entry for "Add Data Export", which can be added somewhere to record the voltage which can be saved and downloaded.  Right-click and Edit the export element to download the data.

closed 2020-09-10 by pfalstad

primebow, 2021-05-18:
Hi, most students are using ecxel diagrams, ploting I-U-curves, for example with diodes. How can get synchronized datas in your example sheet: 
 https://tinyurl.com/yzfqsehm
thanks

pfalstad, 2021-05-18:
You need to use a CCVS to convert the current to voltage, and then have two data export elements:

https://tinyurl.com/yjsvp4je

Click reset, let it run for a while, click stop, and then retrieve both sets of data by right-clicking on each export element, selecting Edit, and downloading the data link.

primebow, 2021-05-19:
Thank you, very much.

 

https://tinyurl.com/yjsvp4je

primebow, 2021-05-25:
Hello Paul,

 

I was able to test your circuit. However, the data is not as easy to process in excel as with a classic dc-sweep. Could there be such a simulation? Set the start and end of the voltage and record the current?

 

Best wishes,

Walter

 

https://tinyurl.com/yjsvp4je

pfalstad, 2021-05-25:
Well you can use a stop trigger to stop the simulation when the voltage source reaches the top of the sweep:

https://tinyurl.com/yhatlgbk

Just click Reset and then Start to do the sweep.

primebow, 2021-05-26:
Thank you very much, it works.

 

Best wishes

Walter

 

https://tinyurl.com/yhatlgbk

primebow, 2021-05-27:
Hi Paul,

 

I didn’t found any help online to scope the amplitude-frequency characteristic in that circuit

 <http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBsBmALAJhQdgWgJwAcYKCFICkIWVNCApgLRhgBQASrccbZHyy9wKGjTT0o0clKxIx0BOwAmPQQLWa6yxgDMAhgFcANgBd2Ac01osNIXzQI6Y9ljAYtwx3XtRatACp9SxA0Yg8bcXDQtBQ-SFcwO2FbB2jUvySsDAS3KI9fDAxBYRosnPYckvVBbP4+BWJCQnj2AHtwEBQPGhyScGg5WgRpPzoINHbO7r8+3CkUIZRoHv8JqYgZ3rIJMEGmZj4RsCwWeYhx0JAASQAdAGcAN3uHgDV2IA> www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBsBmALAJhQdgWgJwAcYKCFICkIWVNCApgLRhgBQASrccbZHyy9wKGjTT0o0clKxIx0BOwAmPQQLWa6yxgDMAhgFcANgBd2Ac01osNIXzQI6Y9ljAYtwx3XtRatACp9SxA0Yg8bcXDQtBQ-SFcwO2FbB2jUvySsDAS3KI9fDAxBYRosnPYckvVBbP4+BWJCQnj2AHtwEBQPGhyScGg5WgRpPzoINHbO7r8+3CkUIZRoHv8JqYgZ3rIJMEGmZj4RsCwWeYhx0JAASQAdAGcAN3uHgDV2IA

 

or describe attenuation over frequency.

 

I hope I don`t bother you to much.

 

Best wishes

Walter

 

https://tinyurl.com/yhatlgbk

pfalstad, 2021-05-27:
There's no good way to do that.  You can use a AC sweep, which varies the frequency over time, but that doesn't give you the response.. You have to calculate it yourself from the output.   Also, even if you could do it, the simulator doesn't model the capacitance of a diode, so you wouldn't get any interesting results.  The frequency response would be flat.