sharpie7/circuitjs1 #462: Modbus TCP

Fvi589, 2020-07-09:
Hi,
 I would like to integrate this COM Modbus TCP, but that does not compile when I put it in a Java elements file. 
What should I do for this to compile?

Example :
import java.io.* ;
import java.net.* ;
import java.util.*;
....
     // set up socket
      Socket es = new Socket("127.0.0.1",502);
      OutputStream os= es.getOutputStream();
      FilterInputStream is = new BufferedInputStream(es.getInputStream());
      byte obuf[] = new byte[261];
      byte ibuf[] = new byte[261];
      // send request
      os.write(obuf, 0, 12);
      // read response
      i = is.read(ibuf, 0, 261);

See : https://paginas.fe.up.pt/~pfs/recursos/plcs/modbus/sources.html

when I try it with javac (test3.java), it works, I communicate with my server.

Please, it's very important for me.
Thanks.

pfalstad, 2020-07-09:
the de and org directories need to go at the same level as "com", under the "src" directory.

not sure about the modbus directory.

Fvi589, 2020-07-10:
I adapted the code that I compiled with javac and it works, even if for the moment I make my frames by hand. I COM in modbus tcp and I have the same frames as easymodbus and I modify the values ​​in the server.
[image]

how can i do that in lushproject please? Even if I have to take over all the frame structures, I do it !!!
[Paul-FVI - ModbusTcp.pdf](https://github.com/sharpie7/circuitjs1/files/4901770/Paul-FVI.-.ModbusTcp.pdf)

pfalstad, 2020-07-10:
You need a try/catch block.  Notice that in test3.java, the whole thing is enclosed in try { ... } catch (Exception e) { ... }

Fvi589, 2020-07-10:
Right 👍 , but now :

[ModbusElm.java.txt](https://github.com/sharpie7/circuitjs1/files/4901882/ModbusElm.java.txt)

[image]
and without write 
[image]
In my opinion, pb with "import", no ?

pfalstad, 2020-07-10:
Unfortunately you don’t have access to the entire java runtime.  Circuitjs gets concerted to javascript using GWT.  java.net is not included in the list of classes supported, because javascript code  is not allowed to just make socket connections wherever it wants.  Hmm...

Fvi589, 2020-07-10:
Damned !!! 
So, it's not possible to com in modbus tcp ? big disappointment !!!

Fvi589, 2020-07-10:
GAME OVER, NEVER !!!! :)
Paul, 
I started from Java sources, without GWT. And it works !!! VICTORY !!!
I saw that it was version V1.6. I guess there have been a lot of changes since 2.2.13. 

please, do you have the V2.2.13 file without gwt? I think no

I will try to wear all the new components step by step. 

But I'm very happy because I don't need gwt for my apps. I will try to continue to wear my new components in the 2 versions.

THANK YOU VERY MUCH !!!!
Franck VIARD

pfalstad, 2020-07-10:
Great!  Sorry, I have not updated the java version in a long time.  The most recent versions all use GWT.

I saw there are some javascript-based modbus interfaces out there, so it might be possible if you want to use the latest version with GWT.  Not sure how they work.

closed 2020-07-30 by Fvi589