sharpie7/circuitjs1 #463: Unable to scroll from Left to right or zoom on specific components

jrogbhp, 2020-07-09:
I have drawn a big schematic, but am unable to scroll from left to right/ up and down or zoom in on specific components. I am only able to zoom in and out of center.

pfalstad, 2020-07-10:
You can drag it around with alt-drag (or cmd-drag on Mac)

jrogbhp, 2020-07-10:
Sorry my mistake. Thank you. Is there a way to change the colour of lines coming from different sources?  eg. make the 24vdc live lines green and the 12vdc system lines appear yellow or another colour?

pfalstad, 2020-07-10:
Unfortunately no.  But if you go into Options->Other Options and set the voltage max color to 24, then the 24V lines will appear light green and the 12V lines will appear darker green.

closed 2020-12-05 by pfalstad