sharpie7/circuitjs1 #465: Export as link giving a broken link

ferrybig, 2020-07-11:
I make an successive approximation ADC in the schematic editor, and wanted to share this with others via an link. (Since telling others to copy and paste a file takes too much effort)

After pressing the export button, it gave this link:

    https://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+0.34903429574618416+50+5+50%0A160+240+128+176+128+0+20+10000000000%0A160+240+192+176+192+0+20+10000000000%0A160+240+320+176+320+0+20+10000000000%0A160+240+256+176+256+0+20+10000000000%0A160+320+160+256+160+0+20+10000000000%0A160+320+288+256+288+0+20+10000000000%0Aw+256+144+240+128+0%0Aw+256+176+240+192+0%0Aw+256+272+240+256+0%0Aw+256+304+240+320+0%0A160+432+224+368+224+0+20+10000000000%0Aw+368+208+320+160+0%0Aw+368+240+320+288+0%0Aw+208+144+208+208+0%0Aw+208+208+208+272+0%0Aw+208+272+208+336+0%0Aw+208+336+208+400+0%0Aw+288+176+288+304+0%0Aw+288+304+288+400+0%0Aw+400+240+400+400+0%0Aw+176+112+128+112+0%0Aw+176+144+128+144+0%0Aw+176+176+128+176+0%0Aw+176+208+128+208+0%0Aw+176+240+128+240+0%0Aw+176+272+128+272+0%0Aw+176+304+128+304+0%0Aw+176+336+128+336+0%0Ab+160+80+444+362+0%0AL+400+400+400+432+2+0+false+5+0%0AL+288+400+288+432+2+0+false+5+0%0AL+208+400+208+432+2+0+false+5+0%0Ax+179+101+433+104+4+24+Input%5Csanalog%5Csmultiplexer%0Ax+126+106+158+109+4+12+ADC0%0Ax+126+138+158+141+4+12+ADC1%0Ax+126+170+158+173+4+12+ADC2%0Ax+126+202+158+205+4+12+ADC3%0Ax+126+234+158+237+4+12+ADC4%0Ax+126+266+158+269+4+12+ADC5%0Ax+126+298+158+301+4+12+ADC6%0Ax+126+330+158+333+4+12+ADC7%0AR+128+240+96+240+0+0+40+5+0+0+0.5%0AR+128+144+96+144+0+0+40+2.5+0+0+0.5%0AR+128+304+96+304+0+0+40+0+0+0+0.5%0A174+16+80+48+128+1+10000+0.5+ADC0+(10k)%0Ag+16+304+16+320+0%0Ag+16+144+16+160+0%0AR+16+80+16+48+0+0+40+5+0+0+0.5%0AR+16+240+16+208+0+0+40+5+0+0+0.5%0Aw+48+272+128+272+0%0Aw+48+112+128+112+0%0A174+16+240+48+288+1+1000+0.5+ADC5+(1k)%0Aw+432+224+480+224+0%0Ax+211+354+231+357+4+12+Bit0%0Ax+297+354+317+357+4+12+Bit1%0Ax+402+356+422+359+4+12+Bit2%0Aw+480+224+480+48+0%0A185+640+320+720+320+0+3%0A164+560+624+608+624+0+3+0+0+0+false+0%0Aw+656+688+672+688+0%0Aw+672+688+672+608+0%0Aw+656+656+704+656+0%0Aw+704+656+704+608+0%0Aw+656+624+736+624+0%0Aw+736+624+736+608+0%0A401+928+48+928+128+1+8%5Cs15%5Cs-15%5Cs1000000%5Cs0%5Cs2.5000003225154632%5Cs100000+0%5Cs20%5Cs10000000000+0%0Ac+688+48+688+112+0+2e-8+2.5000003225154632%0Ag+688+112+688+128+0%0A159+560+48+688+48+0+20+10000000000%0Aw+688+48+944+48+0%0A155+1200+256+1264+256+2+0%0A152+1024+320+1104+320+0+2+0+5%0A150+1120+288+1200+288+0+2+0+5%0Aw+1120+272+1120+256+0%0Aw+1120+256+1200+256+0%0Aw+1104+304+1120+304+0%0Aw+1104+320+1200+320+0%0Aw+1104+304+1104+240+0%0Aw+1104+304+1104+352+0%0A150+944+304+1024+304+0+2+0+5%0A155+1424+256+1488+256+0+0%0Ar+1296+256+1360+256+0+2000%0Aw+1296+320+1296+288+0%0Aw+1296+288+1376+288+0%0Aw+1376+288+1376+256+0%0Aw+1376+256+1424+256+0%0Aw+1360+352+1360+256+0%0Aw+1024+336+1024+352+0%0Aw+1024+336+1024+240+0%0Aw+944+288+944+256+0%0Aw+944+256+1120+256+0%0Aw+912+256+944+256+0%0Aw+928+320+944+320+0%0Aw+928+240+928+320+0%0Aw+928+320+928+352+0%0Aw+1360+240+1360+256+0%0Aw+1360+352+1360+368+0%0Aw+928+432+928+464+0%0Aw+928+352+928+432+0%0Aw+928+432+944+432+0%0Aw+912+368+944+368+0%0Aw+944+368+1120+368+0%0Aw+944+400+944+368+0%0Aw+1024+448+1024+352+0%0Aw+1024+448+1024+464+0%0Aw+1360+464+1360+368+0%0Aw+1376+368+1424+368+0%0Aw+1376+400+1376+368+0%0Aw+1296+400+1376+400+0%0Aw+1296+432+1296+400+0%0Ar+1296+368+1360+368+0+4000%0A155+1424+368+1488+368+0+0%0A150+944+416+1024+416+0+2+0+5%0Aw+1104+416+1104+464+0%0Aw+1104+416+1104+352+0%0Aw+1104+432+1200+432+0%0Aw+1104+416+1120+416+0%0Aw+1120+368+1200+368+0%0Aw+1120+384+1120+368+0%0A150+1120+400+1200+400+0+2+0+5%0A152+1024+432+1104+432+0+2+0+5%0A155+1200+368+1264+368+2+0%0Aw+1360+464+1360+480+0%0Aw+928+544+928+576+0%0Aw+928+464+928+544+0%0Aw+928+544+944+544+0%0Aw+912+480+944+480+0%0Aw+944+480+1120+480+0%0Aw+944+512+944+480+0%0Aw+1024+560+1024+464+0%0Aw+1024+560+1024+576+0%0Aw+1360+576+1360+480+0%0Aw+1376+480+1424+480+0%0Aw+1376+512+1376+480+0%0Aw+1296+512+1376+512+0%0Aw+1296+544+1296+512+0%0Ar+1296+480+1360+480+0+8000%0A150+944+528+1024+528+0+2+0+5%0Aw+1104+528+1104+576+0%0Aw+1104+528+1104+464+0%0Aw+1104+544+1200+544+0%0Aw+1104+528+1120+528+0%0Aw+1120+480+1200+480+0%0Aw+1120+496+1120+480+0%0A150+1120+512+1200+512+0+2+0+5%0A152+1024+544+1104+544+0+2+0+5%0A155+1200+480+1264+480+2+0%0Aw+1360+576+1360+592+0%0Aw+928+656+928+688+0%0Aw+928+576+928+656+0%0Aw+928+656+944+656+0%0Aw+912+592+944+592+0%0Aw+944+592+1120+592+0%0Aw+944+624+944+592+0%0Aw+1024+672+1024+576+0%0Aw+1024+672+1024+688+0%0Aw+1360+688+1360+592+0%0Aw+1376+592+1424+592+0%0Aw+1376+624+1376+592+0%0Aw+1296+624+1376+624+0%0Aw+1296+656+1296+624+0%0Ar+1296+592+1360+592+0+16000%0A155+1424+592+1488+592+0+0%0A150+944+640+1024+640+0+2+0+5%0Aw+1104+640+1104+688+0%0Aw+1104+640+1104+576+0%0Aw+1104+656+1200+656+0%0Aw+1104+640+1120+640+0%0Aw+1120+592+1200+592+0%0Aw+1120+608+1120+592+0%0A150+1120+624+1200+624+0+2+0+5%0A152+1024+656+1104+656+0+2+0+5%0A155+1200+592+1264+592+2+0%0Aw+1360+128+1360+144+0%0Aw+928+208+928+240+0%0Aw+928+128+928+208+0%0Aw+928+208+944+208+0%0Aw+912+144+944+144+0%0Aw+944+144+1120+144+0%0Aw+944+176+944+144+0%0Aw+1024+224+1024+128+0%0Aw+1024+224+1024+240+0%0Aw+1360+240+1360+144+0%0Aw+1376+144+1424+144+0%0Aw+1376+176+1376+144+0%0Aw+1296+176+1376+176+0%0Aw+1296+208+1296+176+0%0Ar+1296+144+1360+144+0+1000%0A155+1424+144+1488+144+0+5%0A150+944+192+1024+192+0+2+0+5%0Aw+1104+192+1104+240+0%0Aw+1104+192+1104+128+0%0Aw+1104+208+1200+208+0%0Aw+1104+192+1120+192+0%0Aw+1120+144+1200+144+0%0Aw+1120+160+1120+144+0%0A150+1120+176+1200+176+0+2+0+5%0A152+1024+208+1104+208+0+2+0+5%0A155+1200+144+1264+144+2+0%0Aw+816+544+816+720+0%0Aw+816+720+1024+720+0%0Aw+1024+720+1024+688+0%0AR+640+320+624+320+0+0+40+5+0+0+0.5%0A155+1424+480+1520+624+0+0%0Aw+1424+176+1424+288+0%0Aw+1424+288+1424+400+0%0Aw+1424+400+1424+512+0%0Aw+1424+512+1424+624+0%0Aw+1424+624+1424+704+0%0Aw+1424+704+832+704+0%0Aw+800+512+832+512+0%0Aw+832+512+832+704+0%0Aw+800+480+848+480+0%0Aw+848+480+848+592+0%0Aw+848+592+912+592+0%0Aw+800+448+864+448+0%0Aw+864+448+864+480+0%0Aw+864+480+912+480+0%0Aw+800+416+864+416+0%0Aw+864+416+864+368+0%0Aw+864+368+912+368+0%0Aw+800+384+848+384+0%0Aw+848+384+848+256+0%0Aw+848+256+912+256+0%0Aw+800+352+832+352+0%0Aw+832+352+832+144+0%0Aw+832+144+912+144+0%0Aw+816+320+816+288+0%0Aw+816+288+816+208+0%0Aw+816+208+624+208+0%0Aw+624+208+624+64+0%0Ar+928+128+848+128+0+100000%0AR+848+128+848+96+0+0+40+5+0+0+0.5%0Aw+1360+128+1360+32+0%0Aw+1360+32+912+32+0%0Ab+1306+96+1408+753+0%0Ax+1322+732+1364+735+4+24+Dac%0Ax+1456+725+1528+728+4+24+Output%0AM+1520+208+1520+192+0+2.5%0AM+1520+320+1520+304+0+2.5%0AM+1520+432+1520+416+0+2.5%0AM+1520+544+1520+528+0+2.5%0AM+1520+656+1520+640+0+2.5%0AR+560+624+496+624+1+2+5000+2.5+2.5+0+0.5%0Aw+912+32+912+48+0%0Aw+560+624+560+720+0%0Aw+560+720+640+720+0%0Aw+512+688+560+688+0%0AI+640+720+720+720+0+0.5+5%0Aw+720+720+720+736+0%0Aw+720+736+1104+736+0%0Aw+1104+736+1104+688+0%0AO+480+224+480+256+1%0AO+688+48+688+16+1%0AO+1360+32+1360+-16+1%0Ar+1360+32+1440+32+0+1000000%0Ag+1440+32+1440+48+0%0Ab+800+256+556+763+0%0Ax+565+281+703+284+4+24+Clock%5Cssource%0Aw+560+48+480+48+0%0A162+1632+656+1632+688+2+default-led+1+0+0+0.01%0A162+1664+656+1664+688+2+default-led+1+0+0+0.01%0A162+1696+656+1696+688+2+default-led+1+0+0+0.01%0A162+1728+656+1728+688+2+default-led+1+0+0+0.01%0A162+1760+656+1760+688+2+default-led+1+0+0+0.01%0Ag+1632+688+1632+704+0%0Ag+1664+688+1664+704+0%0Ag+1696+688+1696+704+0%0Ag+1728+688+1728+704+0%0Ag+1760+688+1760+704+0%0Ar+1520+656+1600+656+0+470%0Ar+1520+544+1600+544+0+470%0Ar+1520+432+1600+432+0+470%0Ar+1520+320+1600+320+0+470%0Ar+1520+208+1600+208+0+470%0Aw+1600+208+1632+208+0%0Aw+1632+208+1632+656+0%0Aw+1600+320+1664+320+0%0Aw+1664+320+1664+656+0%0Aw+1600+432+1696+432+0%0Aw+1696+432+1696+656+0%0Aw+1600+544+1728+544+0%0Aw+1728+544+1728+656+0%0Aw+1600+656+1760+656+0%0Aw+800+320+816+320+0%0Aw+800+544+816+544+0%0Ar+128+208+80+208+0+1000000%0AR+80+208+48+208+0+0+40+0+0+0+0.5%0Ao+56+1+0+4102+9.999999920002695+0.0001+0+2+56+3+Input%0Ao+69+1+0+4102+9.999999919971122+0.0001+1+2+69+3+Sample%5Cs%5Ca%5Cshold%5Cscapacitor%0Ao+234+1+0+4103+9.67741935483871+0.0001+2+2+234+3+DAC%5Csoutput%0Ao+238+1+0+4102+0.0000762939453125+0.0001+3+1+OUT0%0Ao+239+1+0+4102+5+0.0001+4+1+OUT1%0Ao+240+1+0+4102+5+0.0001+5+1+OUT2%0Ao+241+1+0+4102+5+0.0001+6+1+OUT3%0Ao+242+1+0+4102+5+0.0001+7+1+OUT4%0A

After opening this link in a new tab, it gives "www.falstad.com unexpectedly closed the connection.", while I expected it to show my circuit.

Pressing the "short URL" button gives the following error: `Shortner error:`

This is the file that gives problems when exported via link: https://pastebin.com/ZHAdDc2z

pfalstad, 2020-07-11:
When I try the "short URL" button it says "Shortner error:Request-URI Too Large", which it is.  It should give a more reasonable error message though whether or not the short URL button is used.  We have no control over the max URL length.  Someone suggested compressing the URL too so it would fit (#257).

pfalstad, 2020-12-11:
Ok this particular circuit works now, since we're doing compression.  For larger circuits, we do give a warning if the URL is too large.

closed 2020-12-11 by pfalstad