sharpie7/circuitjs1 #466: economy switch schematic leads to error

jkischel, 2020-07-12:
If you're building up an economy switch schematic like descibed under 
https://dein-elektriker-info.de/wp-content/uploads/2018/03/Sparwechselschaltung.png

it leads to errorous output (current comes out ouf nowhere), if you change the switches, simulation stops
[image]

pfalstad, 2020-07-12:
I can't reproduce this, could you send me the circuit?  Go to File->Export as Link...

Felthry, 2020-07-24:
Attempting to draw this circuit gives a matrix error due to the wire loop with no resistance created by the switches in this configuration. If you then click run/stop anyway, this reproduces the bug.

[image]

Flipping either switch restores the simulation to normal working order, and flipping them again gives another matrix error, stopping the simulation as the initial bug report said.

pfalstad, 2020-07-24:
Yes, that's probably it.

sharpie7, 2020-08-02:
Fixed

closed 2020-08-02 by sharpie7