sharpie7/circuitjs1 #468: A possible ripoff

jp-diegidio, 2020-07-18:
FYI, check this out, it's not even given for free:

<https://play.google.com/store/apps/details?id=com.appsmak99.EAR6PRO>

jp-diegidio, 2020-07-18:
And here is the video ad on Facebook, where I have seen it first:

<https://www.facebook.com/whereiselectron/videos/315024803206168/>

pfalstad, 2020-07-18:
Thanks.  Maybe you could download it and post a review telling people that they can just go to my site for free?

jp-diegidio, 2020-07-19:
It is a patent violation of the LGPL licence, not just a marketing problem, isn't it?  I'd rather suggest that, if you care, you report the abuse on both the playstore and facebook.

dr-vij, 2020-07-21:
I posted a review that it is your program. they theft it complettaly. I think you should talk to google play about GNU GPL violation

dr-vij, 2020-07-22:
The guys from their google play account told that you gave them the permission to use sources :)

dr-vij, 2020-07-22:
I think you should go here to stop selling your application 
https://support.google.com/legal/troubleshooter/1114905

ormaaj, 2020-08-24:
> I think you should go here to stop selling your application
> https://support.google.com/legal/troubleshooter/1114905

Selling the app doesn't violate anything. A GPL infringement applies if it's a derivative work and they aren't meeting license obligations, but you'd have to register the copyright (on whatever portion of the work you hold copyright to), hire a lawyer, and file a lawsuit to get an injunction for its removal.

Google likely has DMCA safe harbor here so they're only obIigated to follow that process. I have never seen DMCA process used on a collaborative project so no idea how that works. If you're even allowed to make a claim without sole ownership, it wouldn't be on the entrity of the work. Even @pfalstad probably can't legally use DMCA here unless there's special provision for collaborative works I don't know about. (The fraction of ownership would likely be irrelavent.)

In the end all that would accomplish is GPL complience. Selling the app is totally legal.

jp-diegidio, 2020-08-26:
There is at least the issue of GPL compliance as well as a possible copyright infringement.

I haven't installed the app to double-check for violations (what's in the package, what info appears in the app), but a well documented violation report (from the copyright owner, to a provider such as Google Play or Facebook) might just do the trick, i.e. get the app/page removed.  Here is a story:

https://forum.f-droid.org/t/publish-your-gpl-apps-on-google-play-sooner-rather-than-later/827

dullbananas, 2021-06-03:
This tells what you can do: https://www.gnu.org/licenses/gpl-violation.en.html

closed 2021-12-10 by pfalstad