sharpie7/circuitjs1 #469: Feature request: Improved description of MOSFET Beta Value

JonasForssell, 2020-07-18:
Dear Sir,

First of all, thank you for this fantastic piece of software. What a feat to make it so simple to use and pedagogic!

I have difficulties to understand the beta value for a MOSFET I am currently modelling and the instructions are not specific.
The values in the datasheet are normally specified as min, typical and max. Which value of these are normally used for the beta value?

The Rds(on) is specified for a certain gate Voltage Vgs which I assume is to be put in field 1 and 2 of the sheet.
However, the Threshold (V) is unclear if I should use min, typical or max as input. 

This may be obvious to the expert but I am a beginner and perhaps also a normal user for this software.
Any help appreciated.

Thanks
/Jonas Forssell, Sweden

pfalstad, 2020-07-18:
yes, you should the typical values.  We can make that more clear.

sharpie7, 2020-08-02:
Fixed.

closed 2020-08-02 by sharpie7