sharpie7/circuitjs1 #472: some suggestions for circuitjs

xaez, 2020-07-22:
@pfalstad 

Here some suggestions to extend the "features of circuitjs.

The possibility to create and save an own options-file (offline ver. of circuitjs).

This should be embed into the next update, for the offline ver. of circuitjs:
merging of schematic parts (at this time only possible under the online version).

Some newer models: 
johnson-decade-counter (4017)
bcd-decimal decoder (74154, CMOS:4028)
another kind of OTA (CA3080)

thanks and regards
TSt

pfalstad, 2020-07-22:
Merging of schematics is possible in the offline version.  Load circuit one, select all, copy, load circuit two, paste.

For a bcd decoder, use a regular 4-bit decoder but ignore the extra pins.

There's already a ring counter which looks a lot like the 4017.  It's just missing the clock enable and carry out pin.  Do you need those?

xaez, 2020-07-22:
Hello Paul,

thanks for your informations, the hint (using the online ver. to merging circuit parts works, but I'm not everytime online)

about your 4017 question, here are two circuit-examples using this IC

regards
TSt
https://www.youtube.com/watch?v=SKdD86uP4GU&t=23s

uploaded pieces via youtube.com (Taymur Streng)

last release: "digits & tentacles"
www.taymurstreng.bandcamp.com

www.taymurstreng.de

https://www.mixcloud.com/xaez_2

www.youtube.com/search?q=taymur+streng

https://vimeo.com/search?q=taymur+streng

www.facebook.com/your.tstreng

pfalstad, 2020-07-23:
I didn't see any attachment showing circuit examples using the 4017.

xaez, 2020-07-23:
Here are the examples, note:
[image]
 also the CE (clk inhibit) is an important pin, eg for cascading the 4017 
regards
TSt
[image]
[image]
[image]