sharpie7/circuitjs1 #474: 1N4148 diode voltage drop differs with other program

Shichiya-San, 2020-07-26:
I've constructed a simple circuit, in which the voltage drop differs from other program such as LtSpice/ CircuitLab.
Given that the max AC input voltage is 1.5V @ 1khz, resistor = 3.3k and diode is using 1N4148 model

Circuit using Circuitjs 
>https://imgur.com/a/UVfwewY
The peak negative voltage shows -293.523mV

By using the same circuit on CircuitLab, 
>https://imgur.com/a/sgglWhN
I get peak negative voltage of -524.3mV

Similar results of -524.3mV was obtained by using LtSpice. Is there a configuration problem with my circuit or some settings I haven't adjusted?

pfalstad, 2020-07-26:
Oh no!  That model is completely broken.  Try this one: http://tinyurl.com/y6h3y9a2

Shichiya-San, 2020-07-27:
547.536 might be a bit high, precisely the peak value from LtSpice are between -528~-529mV. I have no idea if the diode varies between application to application but 547.536 are still a bit off the 520~530 range. Is the value correct?

Sorry for the bad English and thank you for replying! Love your work

pfalstad, 2020-07-27:
The datasheet shows a range of possible values.  It seems that a variance of 20-30mV is within that range.  I checked the model against the datasheet and it seemed to match.

You could use this model instead: http://tinyurl.com/yy6lwk2u  I copied it directly from CircuitLab.  I don't know if their model is better or not.

Shichiya-San, 2020-07-27:
> The datasheet shows a range of possible values. It seems that a variance of 20-30mV is within that range. I checked the model against the datasheet and it seemed to match.
>
> You could use this model [...]

Ahhh then I see no problem, thanks for the hotfix! Will the fixed diode be updated for the offline version? I like to run this program but unfortunately I don't always have internet access.

pfalstad, 2020-07-27:
Yes, eventually.

closed 2020-07-27 by Shichiya-San