sharpie7/circuitjs1 #475: Feature-Better SPANISH translation

aruffato, 2020-07-26:
Hello, I try to give you a better translation in SPANISH, so I have attached two files.
The first one contains all the translation I think was more appropiate to the "english version.
[locale_es_V1.0.txt](https://github.com/sharpie7/circuitjs1/files/4978644/locale_es_V1.0.txt)

The next one, contains :
the original translation preceded with "//", and in the line below the "new" translation.  This allow you to watch the difference between.
[locale_es_Modificada-AR.txt](https://github.com/sharpie7/circuitjs1/files/4978645/locale_es_Modificada-AR.txt)

Best regards,

Adrian RUFFATO
adrianruffato@gmail.com

pfalstad, 2020-12-05:
Sorry, I didn't have a chance to take a careful look at this, and didn't think I would be able to tell anything by looking at it anyway, since I don't speak Spanish.  But, someone else submitted a pull request with another improved Spanish translation, and when I looked at that, I happened to notice it corrected some errors in the old translation that were obvious even to me, so I accepted his pull request.  You might want to look at the new translation and see if it still has problems.

closed 2020-12-05 by pfalstad

aruffato, 2020-12-10:
Dear Paul,
Thank you for your answer, I'll take a look to the new translation, best
regards.

Adrian RUFFATO