sharpie7/circuitjs1 #476: Feature- Separate simulation engine from GUI

superlou, 2020-07-27:
I've been trying to find a way to simulate interactive circuit elements like switches and power supply voltages, and circuitjs1 is great. However, I don't need the full GUI (or any GUI). I've been poking around CirSim and it looks like there's a SPICE-like (guessing from "stamp") engine built into the class that also builds the GUI. Is it possible to run circuitjs1's Java application for the real-time simulation, with interaction over a socket or other communication means?

This might be less of a feature and more of an appropriate usage question.

pfalstad, 2020-07-28:
No, you'd have to modify it to do that, or to interface with whatever you need it for.  I don't think the javascript version could do socket communication, but the old java version could.

superlou, 2020-07-29:
That's kinda what I figured. If you don't mind, I may look at extracting out the solver aspects for an open source tool. I haven't found anything similar for running interactive simulations in real-time.

closed 2020-07-29 by superlou