sharpie7/circuitjs1 #477: Editing Sub-circuits.

bhansconnect, 2020-08-01:
I am trying to figure out how to edit a sub-circuit after creating it. For example, if I later realize that a sub-circuit has a bug in it.

According to [the sub-circuit guide](https://www.falstad.com/circuit/subcircuits.html), a sub-circuit needs to be saved so that it can be edited later. I understand that I can update the saved sub-circuit at any time, but how do I import the saved sub-circuit back into may main circuit with the new changes? It seems that when I re-open the main circuit, it still uses the old version of my sub-circuit. This happens even if I run create sub-circuit again.

Thanks for any help and the great tool.

pfalstad, 2020-08-01:
I think you have to paste the sub circuit implementation into your circuit, select it, and then recreate the sub circuit there.  Then you can delete the implementation and save the circuit.

There needs to be a better way to do this...  either it should ask you which implementation you want, or there should be version numbers, or....?

bhansconnect, 2020-08-01:
Ok, that makes sense. It would be amazing if at some point in the future you could open the sub-circuit from the main circuit and just edit it in place.

That being said, another fix that might be easier to implement would be an import sub-circuit button in the main menu, or an import sub-circuit button from the sub-circuit instantiation update menu.

pfalstad, 2020-08-02:
Yes, that would be good, but the sub-circuit as it's stored is not in an editable form.  We could store it that way, I suppose, but it would be bigger.  People like to post circuits as links so that might be a problem.

Oh I see what you mean, add an option to import sub-circuits from another circuit?  So you could load new sub-circuits without changing the existing circuit?  That would be an easy solution.

pfalstad, 2020-08-02:
You can also edit the circuit text file, replacing the sub-circuit definitions (lines starting with ".") with the updated ones.

pfalstad, 2020-12-11:
now you can do File->Import From Text, check "load subcircuits only"

closed 2020-12-11 by pfalstad