sharpie7/circuitjs1 #479: Feature - Dependent Voltage and Current Sources (Diamond)

johnczerwinski, 2020-08-03:
I find circuitjs1 very helpful in my learning of circuit analysis, but cannot find the traditional "diamond" for dependent voltage/current sources.
[image]

pfalstad, 2020-08-03:
Controlled sources are available under "Active Building Blocks" but not the diamond notation, because that would require the controlled source element to be connected to another element and I thought that it would be hard to make a reasonable UI for that.  Instead, controlled sources look like a chip, with the control voltage/current as input(s).

pfalstad, 2022-01-30:
Actually we could have the voltage controlled sources refer to labeled nodes for voltage.  And current controlled sources could refer to ammeters, which could have a label added.