sharpie7/circuitjs1 #481: Generated Links

dvdff, 2020-08-11:
The new form of generated links does not include values - which means it is not possible to update values from outside the browser.

pfalstad, 2020-08-11:
Could you describe in more detail what you are doing, that you need to modify values in the links?  I have some javascript code that generates circuit links but it uses old-style links, which still work.

dvdff, 2020-08-11:
We generate circuit with unique values for students in the class that we teach. Previously with the Falstad links, the values were just a component in the url that we could modify with a code script, but this seems to not be possible with the new urls as these are all letters.

If you have some code to generate these links that would be much appreciated.

pfalstad, 2020-08-11:
Well, if your existing code is generating circuits using old style links, it will still work.

If you need to create new circuits that you can modify with a script, you can generate the old style links by doing File->Export As Text, then replacing all the spaces with "+" and URL encoding the rest (replacing newlines with %0A).

Or, you could get the compression library we are using: https://github.com/pieroxy/lz-string/  And then do File->Export As Text, and compress the result using compressToEncodedURIComponent.  Or take a new-style link, decompress it using decompressFromEncodedURIComponent, modify it and then compress it again.

dvdff, 2020-08-12:
Thanks for your advice on this. We have been able to regenerate these links from the suggestions you provided above, by generating a link from the exported text.

Thanks again.

closed 2020-08-13 by pfalstad