sharpie7/circuitjs1 #486: Electrotech parts request

thorelvin, 2020-08-26:
Im am a teacher for future electricians.

Could you look at implementing relays and contactors.

Being able to build a three phase star delta motor starter would be awesome.

If you could add some prefab motor parts would also be very helpful.

Eksample image: https://qph.fs.quoracdn.net/main-qimg-5fe292f71513861e1d56196f4903bd9c

Also the timer circuit are needed. So a timer gets signal and closes a contactor somewhere else in the circuit, and after a preset time it opens again.

Kind regards,
Thor

pfalstad, 2020-08-26:
We have relays, under passive components.  It looks like the relay component would work for contactors as well.

A timer should be easy enough to make using a subcircuit.  See 555 Timer Chip -> Pulse Sequencer.

flotul, 2020-10-13:
Hi there,

I'm also teaching apprentices in electricity for buildings and appreciate a lot to show them "how things works" thanks to your  simple and great simulator.

Me too, I'm missing "basic" electric parts like pulse relays or timers that don't need to be buildup with NE555 or so like suggested because these young people will never see it like this.

Anyway, thanks a lot for this fantastic tool  :+1: 

Roger

pfalstad, 2020-12-11:
I added a simple time delay relay.

flotul, 2020-12-15:
Thank you Paul ;-)