sharpie7/circuitjs1 #490: Firefox Memory consumption

jeanrenaud, 2020-08-31:
The circuitjs1 applet consumes a never-ending increasing memory space in Firefox 81.0b4 (64 bits). I reach like 3 GB in a couple of minutes, then everything starts to hang, i have to close and reopen the browser (after exporting circuit data before it is too late).

It happens even with the default circuit shown at page loading.

Nothing special happens in the most recent version of MS Edge.  A steady 250 MB.

pfalstad, 2020-08-31:
And it worked fine in earlier releases?  You should report this to the firefox people.

jeanrenaud, 2020-08-31:
I can't remember about previous versions, i was using another browser before, and did not kept history about performance of the applet through time.  Thanks anyway.  Probably some kind of bad garbage collector.

pfalstad, 2020-08-31:
I tried this with 81.0b4 on a mac and didn't notice any memory issues.  I will try on windows.

sharpie7, 2020-12-05:
I use Firefox as my main browser adn there doesn't seem to be an issue currently.

closed 2020-12-05 by sharpie7