sharpie7/circuitjs1 #493: How to show fractional value for resistor.

semiChange, 2020-09-10:
if a resistor vaule > 10k ,  then 10.1k still display 10k.

MASTREX, 2020-09-16:
Yeah! Please fix the code.
Decimal part is considered in calculation but rounded to integer when displayed.
**Only happens when integer is of two or more digits.**

Ex: 
$ 1 0.000005 10.20027730826997 34 5 43
r 176 160 432 160 0 99.5
v 176 304 432 304 0 0 40 5 0 0 0.5
w 176 304 176 160 0
w 432 160 432 304 1
x 288 276 317 279 4 24 5V
o 0 64 0 4099 5 0.2 0 2 0 3

[image]

pfalstad, 2020-09-16:
We were doing this on purpose to reduce clutter.

MASTREX, 2020-09-17:
So just provide a button, slider or a text input to change the float precision.

closed 2020-12-05 by sharpie7