sharpie7/circuitjs1 #494: [Feature Request]: Customizable power sources

Geocube101, 2020-09-14:
Just asking if you can add a component to allow for custom amperage and voltage output
It is very useful in mimicking the output of a transformer in which the voltage could be low while the amperage is very high

I'm trying to mimic the output of a car battery (12 volts at around 600-800 amperes) but there is no way to mimic a high current low voltage output (at least I don't see it)

Thanks

Geocube101, 2020-09-14:
Right now I'm using a resistor to get the desired amperage so this is a very low priority task

pfalstad, 2020-09-14:
I'm not sure what you are asking for here.  You can create a 12V voltage source, and then the current will be whatever it needs to be depending on the load.  If you put a 20 milliohm resistor across it then it will give you 600 A.  If you put a 1kohm resistor across a real battery it won't give you 600 A, it will give you 12 mA.

closed 2020-11-29 by sharpie7