sharpie7/circuitjs1 #495: Feature- A Schmitt input NAND

ThorleifBu, 2020-09-14:
I suggest that if a component (2 input) NAND with Schmitt inputs could be made, then my students (and others) could simulate the correct circuit.

In many of the digital circuits my students make, they use a (Schmitt trigger) NAND gate (2 input) to make an oscillator (in real life they use CD4093)
Basically it's just a normal NAND gate, but both inputs have Schmitt levels.
In your component library the NAND are only traditional types with a single reference voltage for Hi/Lo.
The only gate I could find with Schmitt input, was an inverter. So we use 2 inverters and an OR gate, but ... it's not the same.

pfalstad, 2020-12-15:
Right-click on a NAND, Edit, select "Schmitt Inputs"

https://tinyurl.com/ybvkocmx

I would have responded earlier, but I completely forgot this existed.

closed 2020-12-15 by pfalstad

ThorleifBu, 2020-12-16:
Wow, This is super. My explanation drawings to the students will now match the IC chips they use. :-)
Thank you so much