sharpie7/circuitjs1 #496: Issue - Timing reality

ThorleifBu, 2020-09-14:
If I make a "Real life" small oscillator using a Schmitt inverter (actually a CD4093 NAND), a resistor of 100Kohm and capacitor of 4,7uF, and running at 5V.
Then I get a real life oscillation in the area of 3 Hz.
Simulating however has an initial "power on to first shift" of 2 minutes, and there after a frequency of app. 0.0059 Hz (Nearly 3 seconds for a period)
I can fiddle with the simulation speed, but even the fastest only reaches 0.5Hz, and the oscilloscopes are so fast that they are of no use.
Is there any internal setting that will keep the scope speed nice. but simulate (approximately) the actual oscillator speed.?
**The oscillator circuit:**
$ 3 0.000005 5.023272298708815 64 7 50
183 168 160 296 160 0 0.5 1.66 3.33 5 0
w 296 160 296 104 0
w 168 160 168 104 0
r 296 104 168 104 0 100000
c 168 160 168 248 0 0.0000047 1.012194402940085
g 168 248 168 264 0
368 296 160 360 160 0 0
368 168 160 112 160 0 0
o 7 64 0 4099 1.25 1.6 0 2 7 3
o 6 64 0 4099 5 6.4 1 2 6 3

pfalstad, 2020-09-14:
In Options->Other Options I would set the time step to 150u if you're going to simulate a circuit that slow.  Are you talking about 2 minutes real time?  I'm getting a period of around .6 sec by the simulator clock.

I'm not sure how you're getting 3 Hz from the real circuit..  I'd be interested in seeing a scope view of the NAND input and output if you have one.

ThorleifBu, 2020-09-15:
The physical construction blink the LED at 3.6 times/sec
Oscilloscope connected to the oscillator pin 3 measures a period time of 276ms
Oscilloscope connected to Capacitor measures rise time 155ms and fall time 119ms

Looking at the simulated counter, it tell me that one complete cycle (after initial capacitor charging), start at t=518ms and end at t=1.17sec so the simulator calculates a cycle time of 0,65 seconds = frequency 1.5Hz
However, sitting and observing the simulator, a complete cycle take very close to 3 real physical seconds.

Changing "Options->Other Options" to 1.3ms makes the circuit switch at a seemingly real speed.
But then Scope Scroll speed should also be changed (ie to 500ms/div), and the electrons move very fast.

Ahh, I realise - The question boils down to.
Is there an easy way to switch the circuit times into "physical" time. So that slow sequences will go in real time.?

[image]

pfalstad, 2020-09-15:
No, you just have to experiment with the simulation speed slider and compare the time readout with a real time clock.

I don't know how to read that scope.  What are the high and low voltage values for each graph?

ThorleifBu, 2020-09-15:
Ok, thank you Paul. I will fiddle with the values then.
Yes the scope: I used one of those in my electronics teaching lab, and found a x10 and an x1 probe.
The top trace is the capacitor voltage. Low point is appr. 3.06V  High point is 4.1V  (The upper lazy-T indicate 0V for this trace)
The bottom trace is the digital output from the first NAND gate. Lo=0V Hi=6V DC (text: y1:2V=)  (Lower lazy-T is 0V for this trace)
Timebase is 50ms/div.  The time between the two vertical dotted lines is ^t=276ms

pfalstad, 2020-09-15:
Ok those threshold voltages are higher than what the simulator uses by default.  Also the lower threshold seems a bit higher than what the datasheet says for a CD4096?

Fortunately the simulator lets you configure the threshold voltages.  If I set them to 3.06 and 4.1, and set the output to 6V, then I get a 2.9 Hz output.  http://tinyurl.com/y2wuobra

ThorleifBu, 2020-09-15:
Thank you.
I will play around with those settings. It will be super to show a simulation that, more or less, resemble what my students can see on their scopes.  :-)

closed 2020-11-29 by sharpie7