sharpie7/circuitjs1 #499: Subcircuit

q12a, 2020-09-17:
[image]
I have the standalone for windows version from your website http://www.falstad.com/circuit/
Its a wonderful application! I want to create a subcircuit with it. I see it has the option, but i cannot find where is creating that file and I have no idea how i could inject it into the program. Can you make a quick and dirty tutorial for this problem, please? Thank you.

pfalstad, 2020-09-17:
Here you go: http://www.falstad.com/circuit/subcircuits.html

After clicking "Create Subcircuit" it will create a model in memory.  It doesn't create a file.  You need to use it in a circuit and then save that circuit.

q12a, 2020-09-17:
Thank you very much for your reply.
 But it still doesn't work !
When I am at the [click and drag to create an instance] step, it is not doing anything, not creating anything. 
Have you updated the [Standalone (offline) version] or the [browser version] with your latest compiled version? Ive tried both versions and its the same bug in both.
The latest version I can see for Standalone (offline) version is 6.0.10.
Thank you.

q12a, 2020-09-17:
this is a minor bug, but still a bug.
If i have the background on white, I get this error inside "Edit Subcircuit Model"
[image]
But if i leave it as it was original, on the black background, the problem is remedied: 
[image]
I wish you can color the box in both background situations (not only in black as it is now). 
But its a minor bug i can get over it. 
Im more interested in being able to make the subcircuit work in my program. 
You made a superb software here, i really love it.

pfalstad, 2020-09-17:
Can you try it in the online version, and check the console to see if there's an error in there?  If you tell me what browser you have, I can tell you how to check the console.

q12a, 2020-09-17:
Yes, with pleasure.
I use the latest updated mozilla firefox version: 80.0.1 (64-bit)
I dont know how to use the console you are mentioning.

pfalstad, 2020-09-17:
Here you go: https://developer.mozilla.org/en-US/docs/Tools/Browser_Console

Also please send me the subcircuit you created..  Use File->Export As Link

q12a, 2020-09-17:
Here is my update to your program folder : 
[image]
I made a new folder and name it [saves], inside your program folder. And inside it, i store everything i made so far. I will send you the entire folder as a zip file. Dont get your hopes up, im still learning and testing stuff. 
[saves.zip](https://github.com/sharpie7/circuitjs1/files/5241175/saves.zip)
Here is my 555 circuit im trying to make the subcircuit with it:
[Voltage Inverter.txt](https://github.com/sharpie7/circuitjs1/files/5241183/Voltage.Inverter.txt)
OR 
as you requested, as a link:
http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWK0BMB2MAOFms4CwICckh+ICaFFkAUAO7j7nopMv6RQMhEBsIFJEy8BYMGzpCq+MAKEjmIlJxDkA8gFcALrQAeglGxTjBCZWHIs2CWzwDMmFsLX5lLlA6ch79gfYQ2XwEpSCIfb2CI5xEuGHhICHxofHRLFD57SD4+NEw+ETBaABNowRcA4w8QYoBTADMAQ00AG11GS3Io-h8-bkZKn1IfQN6QnhUuKMGoug7mMcFVWdoAY3ZBNDZOxa4wWoBaQuhMtCI3NHs5TEwsjJ4wNHlVHcnuACcNx7EFt72EtZLKbDN6zcCHET2aC+GGwuFoEojKqQ0ZZFg1BrNNq0IThNHlFHIqDIBJJaBoUgIMAIU6kMCQDDgHg9cTbBas7i40QE7lg+IJIiCoXCkXhJBFRhvDk7DlzcBPeU-cjfD5fCRfBX-OAOYYcqXquX4lVGg0TLZDLqo4ZylWsRVA7hycUK14uHYCZLCtx+XyQIT4WgAGQdetUmDi4BATRaAGdatQ6ABZVzuERUolcFDQBBBh1RUEFYlcaNxhM4hkh9V28NqEAANQAOgAHctUUF9O1+RT1g60IA

pfalstad, 2020-09-17:
The problem is that the Out node isn't connected to anything.  It's connected to the middle of a wire.  See at the bottom right where it says "1 bad connection(s)".  The bad connection is shown in red, but you can't see it because the wire it's connected to is also red.  I'm not sure why it's not reporting an error instead of just silently failing.  I'll fix that.

You should probably get rid of those logic inputs because they might interfere with the circuit in use.

q12a, 2020-09-17:
ok, im looking now into this [Browser Console] addon.
Oh, thank you for your response ! I did everything by the ear, you know? Im discovering your software functionality. Im using it for ... more than 1y ( i think).  It is VERY well done ! I fell in love with it from the first sight. :) 
OHOOOO, it is working now. I can instantiate the subcircuit successfully !
It was because of the logic inputs and the output you mentioned. Removing them and leaving only the labels, did the trick excellent. I even put it to the test in a separate circuit. Its working wonderful ! 
You are an angel. 
To prove im an angel too, ill make a proper tutorial for this functionality, and I want you to insert it in your software after that. I mean, if you are ok with it. Ill make it anyway, for me in the first place. I'll come to you soon with the result. 
Here is the test i made and is working excelent; Im VERY happy now !
[image]
[image]

q12a, 2020-09-17:
Now, on another aspect... i did 'crack" your files a bit. Meaning, I wanted to add my custom page inside the Menu and also start with my custom page. Which i truly love to be able to do it !!! Make or leave intentionally this software to be customized ! Because is good for the soul. Trust me. Especially for a user. 
I managed (in time) to add this page inside the menu:
[image]
And actually start the program with that specific page: 
[image]
BUT... this is the "dirty" menu. 
I wish you can add a tools menu on top. OR... exactly like this, a page with all the components. I dont know. But for me it works. I choose what to use, and delete the rest, and continue building the circuit. Especially when i dont know all the shortcuts, as a beginner ! Now i do and use them.
Im showing my customization. You decide what is best. I like your idea only with shortcuts. Its a very good idea. But a bit... hard to learn in the beginning. What im saying, is to give more options to the user. 
But this is purely aesthetic problem. Its not that critical. 
Thank you so much, you are truly great !

q12a, 2020-09-17:
I finished the promised tutorial. I hope you will like it. 
I made it for me, but if you like it, use it, its free !!! As it is, or make a html help page with the images from that psd file. It's all on layers. 
Until the next time.

q12a, 2020-09-18:
Here are a couple of saves i made today, playing and testing the circuits with the opamp and my subcircuit for the inverter. [opamp.zip](https://github.com/sharpie7/circuitjs1/files/5244298/opamp.zip)
 So far is working exemplary. Truly a fantastic software you made. 
I really dislike the save and load implementation, but i understand why it is like that. I do many mistakes and save a lot... and is a pain in the butt how it is made so far. But is good that it has this option so, ill not complain too much.

pfalstad, 2020-09-18:
Thanks for the tutorial!  It might be a good idea to create the subcircuit instance after creating the subcircuit.  Then everything would be in one file when they save it.

When you say you dislike the save and load implementation, you mean the save implementation, right?

q12a, 2020-09-18:
Sorry, my mistake. The load function is normal and fine. Yes, I was thinking on the save function. I also miss having the shortcut ctrl+s which i mechanically do it, but i remember it is not implemented. I love it is remembering the last folder, both for load and save, you have a 10 for that ! 
And also.. having 2 menus.. is disturbing, i often find myself clicking on the top menu and give a couple of swears.
[image]
Im not sure i understand perfectly what you just said: "might be a good idea to create the subcircuit instance after creating the subcircuit". What? make a drawing for me, or a screenshot to get it better, something got lost in translation there.  
Im sharing my user experience, but in general, it is functional and VERY good mechanics in it, so im easily moving over these little annoying unimplemented bugs. I hope it helps you, my view. You take what you like from me, in the end is your product and I happen to like it very much, but also criticize some little inconveniences. My pure intention is to make it better.
- Did you managed to repair that problem with the superpose red things that didnt give an error? 
- Also the box visibility inside the [Edit subcircuit model] ?
- Also the quick menu on top ? I wish you can have it as a text file, and inside write something like: 
quick components menu= resistor, v+, gnd, diode, bjt, mosfet 
and inside the software, those components being there in the quick menu. I think that is the father of super customization. Also, an option from the program itself (maybe) to customize the quick components menu. You know that game, factorio? they have that awesome quick toolbox in which you can add/remove the most used tools. Im sharing some ideas.. maybe you like them and also implement them. Who knows, maybe im lucky. :) 
I can actually send you a very old software "Croclip" and hopefully inspire you for the top menu. They get it right in an earlier version, simple and to the point. In the version i have now, they got stupid, trying to add too much complexity and forget that is a [quick] toolbar. Anyway, that would be an awesome upgrade. Also you can always make it an option somewhere in settings, like "show quick components menu" checkbox. Nice, no? :)

q12a, 2020-09-19:
I really wish you can insert an LDR in the component list, and a motor.

limitedAtonement, 2020-10-21:
I think this can be closed as the original issue was diagnosed as a bad connection.

pfalstad, 2020-10-21:
The original issue is fixed..  I think the user had some other issues, but those should be filed separately.  I did fix the two menus thing, the offline save, and the subcircuit menu with white background.

closed 2020-10-21 by pfalstad

limitedAtonement, 2020-10-21:
Awesome, thanks!

q12a, 2020-10-21:
Awesome, thanks!