sharpie7/circuitjs1 #500: Massive gigaamp output from doubling opamp amplifier

AlexPoulsen, 2020-09-19:
**Circuit link:** http://tinyurl.com/y5np8ewt
[image]
A voltage doubling opamp fed with a -5V to +5V signal with 0A, outputs a -10V to +10V signal with 1,707,251,400,593,314,816 GA

I'm new to circuits and circuit sim, although I've been working on some decent size circuits in circuitjs for a week or two, so I don't know if this is intentional behavior, but it's a truly massive number, taking 60 bits to represent without including the 9 zeros from the giga prefix. This makes me think that it's possibly an error in the simulator. Additionally, [Wikipedia states that 3 GA is the "Total current in Sol's heliospheric current sheet"](https://en.wikipedia.org/wiki/Orders_of_magnitude_(current)), which again suggests that a single opamp cannot output this many amps.

Shortlink to the circuit is at the top.

pfalstad, 2020-09-19:
So there's a definite bug here, but I'm not sure you realize that all the labeled nodes with the same label are connected to each other.  For example there's two "inv extra gain" nodes, both connected to the output of an op-amp, so what you're doing here is connecting two op-amp outputs to each other.  The simulator maybe shouldn't let you do that, and should act reasonably if you do.  I'll look into why you're getting these huge currents.

Still, you should have all the labeled nodes have unique names unless you mean to connect them.

AlexPoulsen, 2020-09-19:
So I thought all the names for the labeled nodes that I’m using are unique, and in my quick check on my phone, it appears the duplicate is in a bunch of scavenged parts of a previous attempt at this circuit that I haven’t deleted yet.

AlexPoulsen, 2020-09-19:
But I will disconnect them all in the future

AlexPoulsen, 2020-09-20:
ok so I finally tried it again after disconnecting the other labeled nodes and the scopes show 0 amps. thanks!

closed 2021-02-16 by sharpie7