sharpie7/circuitjs1 #501: Feature request: add icons to main menu

SEVA77, 2020-09-20:
Hello. I have a good idea. It would be nice to add icons to the main menu items using webfonts. I did it in my branch [add-icons-to-main-menu](https://github.com/SEVA77/circuitjs1-webapp/tree/add-icons-to-main-menu) for the "File", "Edit" and "Scopes" sections.

[image]
[image]
[image]

Here's a demo: https://seva77.gitlab.io/projects/demo/circuitjs1/2-2-15js/circuitjs.html

I made this webfont in the [Fontello](http://fontello.com/). You can add your icons there and use in your application.

You can see my selected icons [here](https://seva77.gitlab.io/projects/demo/circuitjs1/fontello-77b5f2cc/demo.html). The archive with this font is [here](https://seva77.gitlab.io/projects/demo/circuitjs1/fontello-77b5f2cc.zip).

By default, the icon size is equal to the system font size, but you can change this on [line 149](https://github.com/SEVA77/circuitjs1-webapp/blob/8f63cdd48c872decfa30241b0434f6d571789b76/war/circuitjs.html#L149).

If everything suits you, I can try to create a pull request. I just need to wait for your changes in this repository to avoid conflicts...

pfalstad, 2020-09-21:
Very nice, looks pretty fancy!  Are all the fonts free to use?  Do we need to add attribution in the about box or anything like that?

SEVA77, 2020-09-21:
I think you can just make a link to the file [LICENSE.txt](https://github.com/SEVA77/circuitjs1-webapp/blob/add-icons-to-main-menu/war/font/LICENSE.txt). But this is a must.

Custom icons:
https://www.flaticon.com/free-icon/view-11_32368?term=magnifier%20scale&page=1&position=7
https://www.flaticon.com/free-icon/select-all_335510?term=select%20all&page=1&position=7
https://www.flaticon.com/free-icon/columns_565653?term=column&page=1&position=17
https://www.flaticon.com/free-icon/menu_1828850?term=menu&page=1&position=58

The rest of the icons are taken from other web fonts provided by Fontello. They are all free.
You can also import from config.json (see [archive](https://seva77.gitlab.io/projects/demo/circuitjs1/fontello-77b5f2cc.zip)) on the [Fontello](http://fontello.com/) website and see which icons I've chosen, including custom ones.

SEVA77, 2020-09-21:
I can make this about box so that the link for License.txt opens inside this about box. It would probably be better to create webfontlicense.html instead of License.txt. Then the content inside the about window will also be a separate html file.

pfalstad, 2020-09-21:
It should probably be a separate html file anyway.  It's getting really long.

SEVA77, 2020-09-22:
Ok. I did it.

I placed the content of the about box in a separate html file. You can create links with the target="_self" attribute and the page will load in the aboutbox's iframe.

Changes: https://github.com/SEVA77/circuitjs1-webapp/commit/a43364e95c49c8a3f0886f95d718f28b8dd039b0

and demo: https://seva77.gitlab.io/projects/demo/circuitjs1/2-2-15js/circuitjs.html

[image]

sharpie7, 2020-12-05:
Thanks for helping with this.

closed 2020-12-05 by sharpie7