sharpie7/circuitjs1 #504: [Feature Request] Link two or more circuit elements, to alter features simultaneously

TheRookieNerd, 2020-10-01:
Hey, thank you for this awesome tool, I know you just replied that there's currently no way to do this, but creating a post just in case it gets done in the future. 
For example, in [this circuit](http://tinyurl.com/y4yvpgrw)
It's tedious to alter the parameters of every single pulse gen and the AC source, every time I try to change the firing angle of the SCR. Will it be possible to add a **common slider** for a parameter that is common for one or more components, for example, the frequency and the max voltage of the AC sources?
I really don't know the complications this brings forth with it, excuse me if this is impractical.

pfalstad, 2020-10-01:
Why do you have three separate AC sources, if you want them to be identical?  Why don't you just have one and connect it to all three points?

TheRookieNerd, 2020-10-01:
They are phase shifted by 120 degress

pfalstad, 2020-10-01:
Ah I see, the sliders have different min and max

SEVA77, 2020-10-01:
@pfalstad This may not only apply to AC sources. I remembered that I was getting the [same issue](https://github.com/SEVA77/circuitjs1/issues/7).

There is an idea to create groups of editable parameters in one "Add Sliders" window for setting with one slider. And then users can add an element to a group by checking the box next to this group in the "Add Sliders" window.

Felthry, 2020-10-05:
This might be too complicated, but the idea comes to mind that you might be able to define a field's value as "X" and another one as "2X" and another as "log(X)" or whatever, and then just have a slider for X. That would solve this problem and others of the same sort in a pretty general way.

closed 2022-01-15 by pfalstad