sharpie7/circuitjs1 #508: Convergence failed when using CCVS and VCVS to compute power

Felthry, 2020-10-09:
I was playing around with a TRIAC dimmer circuit, and was curious about the average power dissipated in the TRIAC. It's not possible to view a TRIAC in the scope in circuitjs, so I tried doing this: (apologies for the link instead of embed, something's gone wrong with the upload system)

https://i.imgur.com/9FHd0IP.png

where the CCVS has a formula of "i" and the VCVS has "(a-b)*(c-d)", which should make it output one volt per watt dissipated in the TRIAC (at least, due to current through MT1-MT2. Gate current would also add slightly more dissipation but a pretty tiny amount).

But whenever I try to run the simulation, I get a convergence failed error immediately. I tried adding some grounds to the CCVS and VCVS's outputs, but that didn't fix the problem.

Clearly, something's wrong with how the controlled voltage sources work, but also why is it not possible to view the power dissipation in a TRIAC in the scope anyway? Come to think of it, I don't think you can view the power dissipation of a triode either. Does the scope just hate things that start with "tri"?

pfalstad, 2020-10-09:
Yeah, there's something funny going on there because I get a convergence failure even if the functions are set to a constant, like 0.

I think you can only view 2-terminal devices in the scope because it's not clear which voltage/current to display.  The exception I can think of are MOSFETs (where we just pretend source and drain are the only terminals) and transistors, where I went to extra effort to let you select which terminals you are interested in, and I don't want to do that for TRIACs and triodes.  It wouldn't be that hard to at least treat a TRIAC or triode like a MOSFET, where we just pick 2 terminals and pretend those are the only ones.

Felthry, 2020-10-10:
That's odd; there's definitely something wrong with the VCVS and CCVS then--and probably also the VCCS and CCCS too, if those two are broken.

I think generalizing the BJT code to other elements might be a good idea, so you can look at a MOSFET's Vgs or Vds or a jfet's Ig, or whatever measurement you might need to do on a triac or something.

Transformers also, I think don't have scope code? That would be pretty easy to just let you pick which winding you care about, I would imagine.

closed 2020-12-05 by sharpie7