sharpie7/circuitjs1 #509: Feature request - how to find a specific value of voltage source or resistor?

TeoColuccio, 2020-10-11:
I recently started using this simulator to compare the results of my exercises, and for now it is very useful. In some cases, however, I run into exercises in which I do not know the value of a resistor or a generator and I have to derive it from other information. In fact, in this example, I had to obtain the result of the voltage of the second generator (the one on the right), starting from the power supplied by the other and from the value of the resistors. Doing the calculations by hand, the result was 1V, and entering it in the simulator, all other values match, so it's correct. But is there a way to insert a generator or a resistor of value x to get this value from the simulator itself?

[image]

pfalstad, 2020-10-11:
Unfortunately no.  This seems like it would only be useful for solving homework problems, so it's not likely to happen.

Since this is a linear circuit, the power supplied by the voltage source on the left is a linear function of the voltage of the voltage source on the right.  So you could set the voltage source on the right to 0 V, and get the power supplied on the left side (call it a); set it to 1 V, get the power supplied on the left side (call it b); then use that information to get the correct voltage.

TeoColuccio, 2020-10-11:
Ok, thanks for the tip

closed 2020-12-13 by pfalstad