sharpie7/circuitjs1 #511: Standalone Windows Version Not closing

TonyTWoof56, 2020-10-26:
I and my class mates use this program a lot for our school work and we have found a major problem in the newest offline version. When trying to close the application it will not shut down. The only way to do so is going into task manager and forcing the program to close. Will their be a fix for this soon? 

Also thank you for this amazing tool!

pfalstad, 2020-10-27:
I've seen this before.  It's going to be hard to fix, I'm afraid, because it seems to be a bug in Electron (the app we're using to do the standalone versions).

TonyTWoof56, 2020-10-27:
Thank you for the info I'll be sure to keep that in mind and let my classmates know. I also just wanted to thank ya'll for still working on the program and making it better. My professor gets me to explain how to use the program to a lot of her classes and this new update definitely made saving files a lot easier and simpler to explain.

Lance-at85, 2020-11-15:
Hi,
This is just for your information, I have just installed the Windows standalone version and I too am experiencing this problem.. I have to go into the Windows "Resource Monitor" and end the process to close the program..
Oh, by the way. Thank you very much for supplying the DIY community with this great program..

SEVA77, 2020-11-30:
> I've seen this before. It's going to be hard to fix, I'm afraid, because it seems to be a bug in Electron (the app we're using to do the standalone versions).

I'm not sure but I have an assumption that the program gets stuck at the moment when a warning about unsaved files should appear. Try to remove the option to show this warning.

pfalstad, 2020-12-11:
Wow nice work SEVA77!  I removed that warning and it seems to work.  I thought for sure it would be more complicated than that.

SEVA77, 2020-12-11:
Cool!

closed 2020-12-12 by sharpie7