sharpie7/circuitjs1 #517: HELP - Bistable Multivibrator Using 555 Timer

SM-Fahim, 2020-11-26:
The thing I'm stuck at is the trigger and reset option in the circuit. There is nothing like this option. From example circuits,  so far I've designed the following, but it's not working. 
```
$ 1 0.000005 10.20027730826997 50 5 43
L -32 352 -64 352 0 1 true 5 0
165 32 192 64 192 6 0
R 96 160 96 112 0 0 40 5 0 0 0.5
w 176 224 160 224 0
r 0 224 0 160 0 10000
w 0 160 96 160 0
w 0 224 0 320 0
w 0 320 32 320 0
c -32 352 0 352 0 0.00001 5
c -128 352 -96 352 0 0.00001 5
L -128 352 -160 352 0 1 true 5 0
w 0 352 0 320 0
w 0 352 0 384 0
w 0 160 -96 160 0
r -96 160 -96 240 0 10000
w 32 288 -96 288 0
w -96 240 -96 288 0
w -96 288 -96 352 0
w -96 352 -96 384 0
w -96 384 0 384 0
c 96 352 96 384 0 1e-8 3.333333333333337
w 0 384 96 384 0
w 128 352 128 384 0
w 96 384 128 384 0
g 128 384 128 400 0
368 160 256 224 256 0 0
w 176 224 176 240 0
w 176 240 16 240 0
w 16 240 0 224 0
o 25 64 0 4098 0.0000762939453125 6.4 0 2 25 3

```

pfalstad, 2020-12-11:
I don't understand the question.  You need help designing a circuit?  Or you're having trouble with the simulator?

SM-Fahim, 2020-12-12:
I was trying to build and simulate the circuit Bistable Multivibrator using 555 Timer. The circuit includes trigger and reset option. Which I couldn't make. Any help to complete and simulate the circuit will be highly appreciated.

pfalstad, 2020-12-12:
Is that a circuit you found somewhere on the internet?  where?