sharpie7/circuitjs1 #518: embed the simulation in a PyQt / QML app?

eMPee584, 2020-11-27:
A PyQt app could be styled more flexible and could use system libraries instead of rotten electron..
Especially replacing the current menu with a responsive grouped tabbar would be mighty awesome :rainbow: ..

pfalstad, 2020-11-27:
It would have to be rewritten in python, wouldn't it?

eMPee584, 2020-11-27:
several possibilities I think.. one way would be a qtwebengine embedded view controlled via JS / DOM interaction.. :man_shrugging:

eMPee584, 2020-12-26:
[..]
.. but I ain't gettin eclipse for this now   😯🧟‍♂️ ⏳
could you maybe pleease upload a zipped release of the compiled app? : )

pfalstad, 2020-12-26:
You can download the offline version here: http://www.falstad.com/circuit/offline/

eMPee584, 2020-12-26:
thanks.. yeah cool I had found that in the meantime, will include in our https://schulstick.org/ live distro.. for now as electron app.. I'll have another look into embedding it in a PyQt app.. Best regards!

closed 2021-01-14 by sharpie7