sharpie7/circuitjs1 #522: Bug - Current from nowhere

nietamten, 2020-11-29:
https://tinyurl.com/y3sx3jy2
When the switch is closed and the simulation is restarted, the results are correct but switching twice causes the simulation to be in a bad state.

pfalstad, 2020-11-29:
The problem is that the op-amp is drawing current from ground, but there's no ground in the circuit, so the simulator has to assume a ground node and draw current from there, which looks weird.  If you add a ground somewhere, it will look right.  I'm not sure what to do in this case.  Op-amps probably shouldn't be created in circuits with 2-node voltage sources and no ground.  They have a hidden connection to their positive and negative supply voltages, so they can seemingly create current out of thin air.

nietamten, 2020-11-30:
I don't know much about electronics and op-amps, I'm just a guessing.
I think it works as expected after such changes: https://tinyurl.com/y5z3lrc2

nietamten, 2020-12-06:
Workaround is sufficient for me.

closed 2020-12-06 by nietamten