sharpie7/circuitjs1 #524: Question: bipolar transistor operating modes?

ggeorgovassilis, 2020-11-30:
Hi,
I'm frequently surprised with the simulation accuracy (many thanks for that!), but recently had an issue where the real circuit's behaviour diverged from the simulated behaviour significantly. My question is: which transistor model is the simulation based on and which operation modes (A,B,C) does the simulation implement?

Some background:
The circuit in question is a common emitter with NPN which is used to amplify a sine wave. The issue:

1. the simulated circuit is much more sensitive than the real one. In the simulation, R1 = 100KΩ, which produces a clean, high-amplitude output signal. The real circuit requires lower values for R1.
2. the simulated circuit is distortion-free; the real circuit caps the tops of the wave signal at an amplitude of about 6V.

https://lushprojects.com/circuitjs/circuitjs.html?ctz=CQAgjOCmC0Cc4AYB0AmBCUHZMGYEA4UA2WWTEAVgUpABYcAoAFxH2qxXCPc0+omhhaqWmhwVuYIlJz4IyCqRyYwYHCirL6EMOgYAnLuwlHT-dAgYAlU1P7dwYPiGq17LjwoYBzUzhxEpm7UloZsIBys9iGICLB6AO5Rtg52LgbJkeH+gdTxick5fgHpSWlFaSj4+OkBNZUmKCi0ESYhDADGmbwg0OoRPfwwqr04qAiYpJj4AbDNYPi0tNa9qpyRgg6R-LEWeSA4nkgUGX3rPYJOA866ewwAHr2wECh4vdIR-nQRLVZgD+BSBEwBQIih4CgFt9miArMtHoFITUSk0IC0YVYUAC+gcYVhceR0b9lgB7EDwIgtVxxGpoFxIeQRckHBhAA

pfalstad, 2020-11-30:
It's based on an Ebers-Moll transistor model which incorporates all the operation modes.  You can edit the transistor (right click, edit) to change the transistor beta, or current gain, which will affect the transistor behavior.  You mentioned that the real circuit is less sensitive, so maybe the beta should be reduced.  If I use a beta of 50, that reduces the amplitude by half.

The transistor model doesn't include any capacitance (Miller effect) which would reduce the gain at high frequencies, but I don't think that would be a factor at 1 Mhz.

I can't think of why the circuit would be clipping the waveform at 6V.

ggeorgovassilis, 2020-11-30:
Thank you Paul.

closed 2020-11-30 by ggeorgovassilis