sharpie7/circuitjs1 #525: Half-wave rectifier question

J0RJO, 2020-11-30:
Hello,

i want to ask something about half-wave rectifier simulation circuit.

why Vd is 4,5V?

i think the correct answer should be:

Vr=5V
Vrms=0,707*VP=3,5V

If we consider and V draw in diode...should be around 2,7V.

I would be glad if you would answer me.

Regards,
Jorjo

pfalstad, 2020-11-30:
Well, the input is 5V, and there's about a 550mV voltage drop across the diode at that current (about 7mA), so that leaves 4.45V.

Why is the rms voltage relevant here?  4.45 V is the peak voltage.

J0RJO, 2020-11-30:
Thank you for your response.

in the attached picture, my mouse is over resistor.

i don't see the Vdc. I see only Vmax, but after diode we have Vdc, right?

(i also create that circuit in Multisim, to try help you understand what i am trying to say)

Regards

________________________________
Από: Paul Falstad <notifications@github.com>
Στάλθηκε: Τρίτη, 1 Δεκεμβρίου 2020 12:14 πμ
Προς: sharpie7/circuitjs1 <circuitjs1@noreply.github.com>
Κοιν.: J0RJO <jorjo.satiros@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [sharpie7/circuitjs1] Half-wave rectifier question (#525)

Well, the input is 5V, and there's about a 550mV voltage drop across the diode at that current (about 7mA), so that leaves 4.45V.

Why is the rms voltage relevant here? 4.45 V is the peak voltage.

pfalstad, 2020-11-30:
Your picture didn't get attached.  If you want to attach a picture, you have to go to github and do it.  You can't reply with an attachment in email..  The attachment will be ignored.

J0RJO, 2020-11-30:
i upload them in ImgBB.

if this dont work, i will reply in my post in github.

https://ibb.co/gZyrRs0<https://ibb.co/gZyrRs0https://ibb.co/HPfJVSb>

https://ibb.co/HPfJVSb<https://ibb.co/gZyrRs0https://ibb.co/HPfJVSb>

________________________________
Από: Paul Falstad <notifications@github.com>
Στάλθηκε: Τρίτη, 1 Δεκεμβρίου 2020 12:28 πμ
Προς: sharpie7/circuitjs1 <circuitjs1@noreply.github.com>
Κοιν.: J0RJO <jorjo.satiros@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [sharpie7/circuitjs1] Half-wave rectifier question (#525)

Your picture didn't get attached. If you want to attach a picture, you have to go to github and do it. You can't reply with an attachment in email.. The attachment will be ignored.

J0RJO, 2020-11-30:
broken link, sorry

https://ibb.co/gZyrRs0
[https://i.ibb.co/qYBDrbH/2.jpg]<https://ibb.co/gZyrRs0>
2<https://ibb.co/gZyrRs0>
Image 2 hosted in ImgBB
ibb.co

https://ibb.co/HPfJVSb
[https://i.ibb.co/T0xCkf7/1.png]<https://ibb.co/HPfJVSb>
1<https://ibb.co/HPfJVSb>
Image 1 hosted in ImgBB
ibb.co

________________________________
Από: Paul Falstad <notifications@github.com>
Στάλθηκε: Τρίτη, 1 Δεκεμβρίου 2020 12:28 πμ
Προς: sharpie7/circuitjs1 <circuitjs1@noreply.github.com>
Κοιν.: J0RJO <jorjo.satiros@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [sharpie7/circuitjs1] Half-wave rectifier question (#525)

Your picture didn't get attached. If you want to attach a picture, you have to go to github and do it. You can't reply with an attachment in email.. The attachment will be ignored.

pfalstad, 2020-11-30:
Looks about the same..  Multisim looks like it peaks at about 4.5 V according to the scope.

I don't know why the multimeter is reading 1.3 V..  That doesn't seem to match what the scope is reading.  But I don't know if that's instantaneous or an average.

J0RJO, 2020-11-30:
Now, i understand. In resistor, you keep measure the Vpeak. No the real voltage.

Just in case, i explain myself:

We have 5Vmax (Vpeak)

After Diode we have Vdc.

1) Vdc=Vmax/3,14=5/3.14=1.59V

2) Vdc=Vrms*0.45= **3,5V*0.45=1.57V

[**Vrms=Vmax*0.707=3,5V]

3) Vdc=0.318*Vmax=1.59V

with these 3 formulas, we can find Vdc (the voltage we can measure with a voltometer-the real voltage).

Thank you for your reply.

________________________________
Από: Paul Falstad <notifications@github.com>
Στάλθηκε: Τρίτη, 1 Δεκεμβρίου 2020 12:37 πμ
Προς: sharpie7/circuitjs1 <circuitjs1@noreply.github.com>
Κοιν.: J0RJO <jorjo.satiros@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [sharpie7/circuitjs1] Half-wave rectifier question (#525)

Looks about the same.. Multisim looks like it peaks at about 4.5 V according to the scope.

I don't know why the multimeter is reading 1.3 V.. That doesn't seem to match what the scope is reading. But I don't know if that's instantaneous or an average.

pfalstad, 2020-11-30:
The Vd is actually the instantaneous voltage across the diode at that moment, not Vpeak.  For Vpeak you need to look at the scope view.

J0RJO, 2020-11-30:
Yeah, you just want to show the 0,7v drop in diode.

i was caring about Vdc (the real voltage which we will take parallel of the resistor)

________________________________
Από: Paul Falstad <notifications@github.com>
Στάλθηκε: Τρίτη, 1 Δεκεμβρίου 2020 12:50 πμ
Προς: sharpie7/circuitjs1 <circuitjs1@noreply.github.com>
Κοιν.: J0RJO <jorjo.satiros@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [sharpie7/circuitjs1] Half-wave rectifier question (#525)

The Vd is actually the instantaneous voltage across the diode at that moment, not Vpeak. For Vpeak you need to look at the scope view.

closed 2020-12-05 by pfalstad