sharpie7/circuitjs1 #529: Feature Request: Interact only mode

EmergencyTemporalShift, 2020-12-05:
There should be an interact-only mode.
In this mode, you could toggle switches or potentiometers, but not accidentally drag components around or open any menus by clicking and holding or double-clicking. (This is especially useful for momentary switches) This might be helpful if you have a specific circuit you want to simulate and don't want to change it (Like maybe a classroom setting).

To indicate that the circuit is in an interact-only mode, you could put a message in the top left that says editing disabled. Hovering over this message would tell you that you can disable it in the settings.

Copied and slightly edited from https://github.com/sharpie7/circuitjs1/issues/68

pfalstad, 2020-12-05:
This is also #467.  So no editing with right mouse menus, no adding/removing scopes, no changing scope parameters, no mouse wheel to change scope speed, etc.?

pfalstad, 2020-12-11:
added to 2.3.0

closed 2020-12-11 by pfalstad