sharpie7/circuitjs1 #530: Bug: Mosfet matrix error and undo/export breakage

EmergencyTemporalShift, 2020-12-05:
If you make a p or n-type MOSFET and edit it so that "Show Bulk", "Simulate Body Diode", and "Body Terminal" are on, then disable "Show Bulk", it will give a matrix error and break in the way described below.
Turning off "Simulate Body Diode" will also give a matrix error.
Also, with all options off, turning on "Show Bulk" will give a matrix error.

In the first case, (disabling "Show Bulk",) The node labeled "B" does not move when the MOSFET is moved. After you edit the circuit, hitting undo and exporting a link/text no longer loads the circuit properly, showing a white screen. Whether you edit the circuit or not, if you do not press undo before exporting, you can still load the circuit, but are unable to edit it.
Working:
```
$ 1 0.000005 10.20027730826997 50 5 50
f 272 256 272 320 96 1.5 0.02
```
Breaking with undo:
```
$ 1 0.000005 10.20027730826997 50 5 50
f 304 256 304 320 116 1.5 0.02
```
Breaking without undo:
```
$ 1 0.000005 10.20027730826997 50 5 50
f 336 256 336 320 116 1.5 0.02
```

pfalstad, 2020-12-05:
As a workaround, it looks like it will work if you turn off "body terminal" before turning off Show Bulk or Simulate Body Diode.

pfalstad, 2020-12-11:
fixed in 2.3.0

closed 2020-12-11 by pfalstad