sharpie7/circuitjs1 #534: Feature- Show chip variants

EmergencyTemporalShift, 2020-12-13:
In Sequential Logic/LED Flasher the circuit uses a ring counter without a ce input. This type doesn't seem like it's able to be drawn via menu. I was able to make several hidden types by editing the import text. If these variants are already implemented, they should be accessible by some menu.
```
$ 17 0.000005 2.1831051418620846 50 5 50
163 -32 32 -16 32 0 10 5 0 0 0 0 0 0 0 0 0
163 -32 128 -16 128 1 10 5 0 0 0 0 0 0 0 0 0
163 -32 224 -16 224 2 10 5 0 0 0 0 0 0 0 0 0
163 -32 336 -16 336 3 10 5 0 0 0 0 0 0 0 0 0
```
Here's a link: https://tinyurl.com/yyabsm88

pfalstad, 2020-12-13:
I didn't bother making a UI for turning off the CE input because I didn't think it was necessary..   You can just ignore the CE input, leaving the pin unconnected, and it will work the same.

The other two variants are just small grid versions, which you can get by turning on Options->Small Grid and creating a ring counter or other chip.

EmergencyTemporalShift, 2020-12-13:
Then why does it have a variant at all? Why is the variant used?

pfalstad, 2020-12-13:
I added the CE pin but wanted to make sure that existing circuits wouldn't be affected.  Maybe someone had a connection where the CE pin is now.  So I made the CE pin only for new ring counters.

closed 2021-02-16 by sharpie7